xreef/LoRa_E32_Series_Library #1: How to Send Message use JSON Format ?

taurushappy, 2019-12-16:
How to Send Message use JSON Format ? 
     Hello , Can you help me ?  How to solve this  Question ?

xreef, 2019-12-16:
Hi taurushappy,
you can use ArduinoJSON library, (I'm going to publish a post on how to use It).

You can 
```
  DynamicJsonDocument doc(512);

  // Set the values in the document
  doc["energyLifetime"] = 21698620;
  doc["energyYearly"] = 1363005;

String buf;
serializeJson(doc, buf);

ResponseStatus rs = e32ttl.sendMessage(buf);
```

Remember you have a size limit on message size of the lora E32.

The code is untested, but I think It can be usefully to understand.

Tell me if It's ok.

Bye Renzo

taurushappy, 2019-12-16:
HI !
   first, thanks  for your help !
   I try this ,the code can work ,but I don't know it have some problem in my test ,it only work one time,send  Message one time ,then stoped!
    this  is  my test code : 

 #include "Arduino.h"
#include "LoRa_E32.h"

#include <DHT.h&gt;
#include <ArduinoJson.h&gt;

#define DHTPIN 4
#define DHTTYPE DHT11

DHT dht(DHTPIN, DHTTYPE);

DynamicJsonDocument doc(512);

// ---------- esp8266 pins --------------
//LoRa_E32 e32ttl(D2, D3, D5, D7, D6);
//LoRa_E32 e32ttl(D2, D3); // Config without connect AUX and M0 M1

//#include <SoftwareSerial.h&gt;
//SoftwareSerial mySerial(D2, D3); // RX, TX
//LoRa_E32 e32ttl(&amp;mySerial, D5, D7, D6);
// -------------------------------------

// ---------- Arduino pins --------------
//LoRa_E32 e32ttl(2, 3, 5, 7, 6);
LoRa_E32 e32ttl100(2, 3); // Config without connect AUX and M0 M1

//#include <SoftwareSerial.h&gt;
//SoftwareSerial mySerial(2, 3); // RX, TX
//LoRa_E32 e32ttl(&amp;mySerial, 5, 7, 6);
// -------------------------------------

void setup() {
  
  Serial.begin(9600);
  delay(500);

  // Startup all pins and UART
  e32ttl100.begin();
  dht.begin();

  Serial.println("Hi, I'm going to send message!");

  // Send message
  ResponseStatus rs = e32ttl100.sendMessage("Hello, Taurus!");
  // Check If there is some problem of succesfully send
  Serial.println(rs.getResponseDescription());
}

int i = 0;

void loop() {

  float h = dht.readHumidity();
  // Read temperature as Celsius (the default)
  float t = dht.readTemperature();
  // Read temperature as Fahrenheit (isFahrenheit = true)
  float f = dht.readTemperature(true);

  // Check if any reads failed and exit early (to try again).
  if (isnan(h) || isnan(t) || isnan(f)) {
    Serial.println(F("Failed to read from DHT sensor!"));
    return;
  }

  // Compute heat index in Fahrenheit (the default)
  float hif = dht.computeHeatIndex(f, h);
  // Compute heat index in Celsius (isFahreheit = false)
  float hic = dht.computeHeatIndex(t, h, false);

  Serial.print(F("Humidity: "));
  Serial.print(h);
  Serial.print(F("%  Temperature: "));
  Serial.print(t);
  Serial.print(F("°C "));
  Serial.print(f);
  Serial.print(F("°F  Heat index: "));
  Serial.print(hic);
  Serial.print(F("°C "));
  Serial.print(hif);
  Serial.println(F("°F"));

  
 
  delay(1000);
    

   doc["temperature"]=t;
   doc["humidity"]=h;
   String buf;
   serializeJson(doc,buf);
   ResponseStatus rs = e32ttl100.sendMessage(buf);

  
  Serial.println(rs.getResponseDescription());

  delay(1000);

}

xreef, 2019-12-16:
Hi I check your sketch and seems quite ok.

But I think you must add AUX pin or put a pull up resistor on It, with no AUX not work for me, but with aux I have a correct send.

Now I write fast sketch and not fully tested. (you must set correct string size)

Not use directry String because json library do something strange on It.

`
#include "Arduino.h"
#include "LoRa_E32.h"

#include <DHT.h>
#include <ArduinoJson.h>

#define DHTPIN 4
#define DHTTYPE DHT11
int freeRame () {
#ifdef ESP8266
	return Esp.getFreeHeap();
#else
	extern int __heap_start, *__brkval;
	int v;
	return (int) &v - (__brkval == 0 ? (int) &__heap_start : (int) __brkval);
#endif // ESP8266
}

DHT dht(DHTPIN, DHTTYPE);

DynamicJsonDocument doc(512);

// ---------- esp8266 pins --------------
//LoRa_E32 e32ttl(D2, D3, D5, D7, D6);
//LoRa_E32 e32ttl(D2, D3); // Config without connect AUX and M0 M1

//#include <SoftwareSerial.h>
//SoftwareSerial mySerial(D2, D3); // RX, TX
//LoRa_E32 e32ttl(&mySerial, D5, D7, D6);
// -------------------------------------

// ---------- Arduino pins --------------
//LoRa_E32 e32ttl(2, 3, 5, 7, 6);
LoRa_E32 e32ttl100(4, 5, 3); // Config without connect AUX and M0 M1

//#include <SoftwareSerial.h>
//SoftwareSerial mySerial(2, 3); // RX, TX
//LoRa_E32 e32ttl(&mySerial, 5, 7, 6);
// -------------------------------------
void printParameters(struct Configuration configuration) {
	Serial.println("----------------------------------------");

	Serial.print(F("HEAD BIN: "));  Serial.print(configuration.HEAD, BIN);Serial.print(" ");Serial.print(configuration.HEAD, DEC);Serial.print(" ");Serial.println(configuration.HEAD, HEX);
	Serial.println(F(" "));
	Serial.print(F("AddH BIN: "));  Serial.println(configuration.ADDH, BIN);
	Serial.print(F("AddL BIN: "));  Serial.println(configuration.ADDL, BIN);
	Serial.print(F("Chan BIN: "));  Serial.print(configuration.CHAN, DEC); Serial.print(" -> "); Serial.println(configuration.getChannelDescription());
	Serial.println(F(" "));
	Serial.print(F("SpeedParityBit BIN    : "));  Serial.print(configuration.SPED.uartParity, BIN);Serial.print(" -> "); Serial.println(configuration.SPED.getUARTParityDescription());
	Serial.print(F("SpeedUARTDataRate BIN : "));  Serial.print(configuration.SPED.uartBaudRate, BIN);Serial.print(" -> "); Serial.println(configuration.SPED.getUARTBaudRate());
	Serial.print(F("SpeedAirDataRate BIN  : "));  Serial.print(configuration.SPED.airDataRate, BIN);Serial.print(" -> "); Serial.println(configuration.SPED.getAirDataRate());

	Serial.print(F("OptionTrans BIN       : "));  Serial.print(configuration.OPTION.fixedTransmission, BIN);Serial.print(" -> "); Serial.println(configuration.OPTION.getFixedTransmissionDescription());
	Serial.print(F("OptionPullup BIN      : "));  Serial.print(configuration.OPTION.ioDriveMode, BIN);Serial.print(" -> "); Serial.println(configuration.OPTION.getIODroveModeDescription());
	Serial.print(F("OptionWakeup BIN      : "));  Serial.print(configuration.OPTION.wirelessWakeupTime, BIN);Serial.print(" -> "); Serial.println(configuration.OPTION.getWirelessWakeUPTimeDescription());
	Serial.print(F("OptionFEC BIN         : "));  Serial.print(configuration.OPTION.fec, BIN);Serial.print(" -> "); Serial.println(configuration.OPTION.getFECDescription());
	Serial.print(F("OptionPower BIN       : "));  Serial.print(configuration.OPTION.transmissionPower, BIN);Serial.print(" -> "); Serial.println(configuration.OPTION.getTransmissionPowerDescription());

	Serial.println("----------------------------------------");

}
[... 76 lines cut ...]
	Serial.print(F("°F Heat index: "));
	Serial.print(hic);
	Serial.print(F("°C "));
	Serial.print(hif);
	Serial.println(F("°F"));
	Serial.println(freeRame());
	delay(1000);

	doc["temperature"] = t;
	doc["humidity"] = h;
	Serial.println(freeRame());
	char buf[31];
	serializeJson(doc, buf, 31);
	Serial.println(buf);
	Serial.println(doc.size());
	ResponseStatus rs = e32ttl100.sendMessage(buf, 31);
	Serial.println(freeRame());
	Serial.println(rs.getResponseDescription());
	Serial.println(freeRame());

	delay(2000);

}

`

xreef, 2019-12-28:
I tested and works correctly, I close issue but I wait feedback.
Thanks and bye Renzo

closed 2019-12-28 by xreef

taurushappy, 2020-02-10:
Thanks  your help solve the problem with send message use json !  I am a Chinese student，and I want to know :
1.  how to buy  your create arduino shield  without use 4.7k resister when use LoRa E32? and how much the your create arduino shield  ?
2. how to  find the  passage  about  "LoRa E32 device for Arduino, esp32 or esp8266: real example ad WeMos D1 shield"

xreef, 2020-02-10:
Hi taurushappy,
Why don't you want to use 4.7k resistors? You can try with similar different value, but datasheet advise 4.7k.
The price isn't relevant, I don't sell the shield, I give the possibility to buy from productor of PCB the PCB with my schema.
I add the link here on github, but It exist already on my site.
Bye Renzo

taurushappy, 2020-02-10:
Thanks !
    If  I want  to  make  data terminal unit   product in lot project , i couldn't use  breadboard , it is too large! so I need small and convenience  pcb instead !

xreef, 2020-02-10:
Ok, if you need some help for your project ask me.
Bye Renzo