xreef/LoRa_E32_Series_Library #2: frozen at ResponseContainer in example sendReceiveTransparentTransmissionMessage

fransolomon, 2019-12-24:
I have tried the example of transparent transmission between two modules. One has this example program that I attached and is configured and wired as shown in the readme of the library and the other is simply connected in transparent mode to the USB port of my PC.
The two E32 modules are connected to arduino mega 2560.

In the code example that I have passed you, the arduino is frozen in the code line:
Serial.println (rs.getResponseDescription ());
If I comment on this line, the program continues and freezes again on this line.
ResponseContainer rc = e32ttl100.receiveMessage ();

I don't know if you can give me a hand, because I'm sure I have something wrong but I don't know where to take it.
Regards, and thank you very much.

#include "Arduino.h"
#include "LoRa_E32.h"

SoftwareSerial mySerial(11, 10);
LoRa_E32 e32ttl100(&mySerial,5, 7, 6);

void setup() {
  Serial.begin(9600);
  delay(500);

  // Startup all pins and UART
  e32ttl100.begin();

  Serial.println("Hi, I'm going to send message!"); 
  // Send message
  ResponseStatus rs = e32ttl100.sendMessage("Hello, world?");  // OK The message is received on the other device
// Check If there is some problem of succesfully send
Serial.println(rs.getResponseDescription());
}

void loop() {
  // If something available
    Serial.println("continuity test point"); //This instruction is not executed unless you comment on the previous line "Serial.println(rs.getResponseDescription());"
  if (e32ttl100.available()>1) {

    // read the String message
  ResponseContainer rc = e32ttl100.receiveMessage();
   Serial.println(rc.data);
  // Is something goes wrong print error
  if (rc.status.code!=1){
    rc.status.getResponseDescription();
  }else{
    // Print the data received
    Serial.println(rc.data);
 }
  }
  if (Serial.available()) {
    String input = Serial.readString();
    e32ttl100.sendMessage(input);
  }
}

xreef, 2019-12-24:
Hi, I can't check the sketch now, but please refer to my site, i update all the schema. 
In the next few days I check all with my Arduino mega.
Thanks to write, and please wait.
Bye Renzo

fransolomon, 2019-12-24:
Thanks for checking my problem. I will be waiting and if I find out something else I will tell you. Greetings

xreef, 2019-12-24:
Hi fransolomon,
I tested your sketch with 2 Arduino UNO, and It's work correctly, I only add the correct configuration, please try It.

```
#include "Arduino.h"
#include "LoRa_E32.h"

// SoftwareSerial mySerial(11, 10);
// LoRa_E32 e32ttl100(&mySerial,5, 7, 6);

SoftwareSerial mySerial(4, 5);
LoRa_E32 e32ttl100(&mySerial, 3, 7, 6);
//LoRa_E32 e32ttl100(4, 5, 3, 7, 6);

void printParameters(struct Configuration configuration) ;

void setup() {
	Serial.begin(9600);
	delay(500);

// Startup all pins and UART
	e32ttl100.begin();

	ResponseStructContainer c;
	c = e32ttl100.getConfiguration();
	Configuration configuration = *(Configuration*) c.data;
	configuration.ADDL = 3;
	configuration.ADDH = 0;
	configuration.CHAN = 0x04;
	configuration.OPTION.fixedTransmission = FT_TRANSPARENT_TRANSMISSION;
	configuration.OPTION.wirelessWakeupTime = WAKE_UP_250;

    configuration.OPTION.fec = FEC_1_ON;
    configuration.OPTION.ioDriveMode = IO_D_MODE_PUSH_PULLS_PULL_UPS;
    configuration.OPTION.transmissionPower = POWER_20;

    configuration.SPED.airDataRate = AIR_DATA_RATE_010_24;
    configuration.SPED.uartBaudRate = UART_BPS_9600;
    configuration.SPED.uartParity = MODE_00_8N1;

    e32ttl100.setConfiguration(configuration, WRITE_CFG_PWR_DWN_SAVE);
	printParameters(configuration);

	Serial.println("Hi, I'm going to send message!");
// Send message
	ResponseStatus rs = e32ttl100.sendMessage("Hello, world?"); // OK The message is received on the other device
// Check If there is some problem of succesfully send
	Serial.println(rs.getResponseDescription());
}

void loop() {
// If something available
//	Serial.println("continuity test point"); //This instruction is not executed unless you comment on the previous line "Serial.println(rs.getResponseDescription());"
	if (e32ttl100.available() > 1) {
		ResponseContainer rc = e32ttl100.receiveMessage();
		Serial.println(rc.data);
// Is something goes wrong print error
		if (rc.status.code != 1) {
			rc.status.getResponseDescription();
		} else {
// Print the data received
			Serial.println(rc.data);
		}
	}
	if (Serial.available()) {
		String input = Serial.readString();
		e32ttl100.sendMessage(input);
	}
}
void printParameters(struct Configuration configuration) {
	Serial.println("----------------------------------------");

	Serial.print(F("HEAD : "));  Serial.print(configuration.HEAD, BIN);Serial.print(" ");Serial.print(configuration.HEAD, DEC);Serial.print(" ");Serial.println(configuration.HEAD, HEX);
	Serial.println(F(" "));
	Serial.print(F("AddH : "));  Serial.println(configuration.ADDH, DEC);
	Serial.print(F("AddL : "));  Serial.println(configuration.ADDL, DEC);
	Serial.print(F("Chan : "));  Serial.print(configuration.CHAN, DEC); Serial.print(" -> "); Serial.println(configuration.getChannelDescription());
	Serial.println(F(" "));
	Serial.print(F("SpeedParityBit     : "));  Serial.print(configuration.SPED.uartParity, BIN);Serial.print(" -> "); Serial.println(configuration.SPED.getUARTParityDescription());
	Serial.print(F("SpeedUARTDatte  : "));  Serial.print(configuration.SPED.uartBaudRate, BIN);Serial.print(" -> "); Serial.println(configuration.SPED.getUARTBaudRate());
	Serial.print(F("SpeedAirDataRate   : "));  Serial.print(configuration.SPED.airDataRate, BIN);Serial.print(" -> "); Serial.println(configuration.SPED.getAirDataRate());

	Serial.print(F("OptionTrans        : "));  Serial.print(configuration.OPTION.fixedTransmission, BIN);Serial.print(" -> "); Serial.println(configuration.OPTION.getFixedTransmissionDescription());
	Serial.print(F("OptionPullup       : "));  Serial.print(configuration.OPTION.ioDriveMode, BIN);Serial.print(" -> "); Serial.println(configuration.OPTION.getIODroveModeDescription());
	Serial.print(F("OptionWakeup       : "));  Serial.print(configuration.OPTION.wirelessWakeupTime, BIN);Serial.print(" -> "); Serial.println(configuration.OPTION.getWirelessWakeUPTimeDescription());
	Serial.print(F("OptionFEC          : "));  Serial.print(configuration.OPTION.fec, BIN);Serial.print(" -> "); Serial.println(configuration.OPTION.getFECDescription());
	Serial.print(F("OptionPower        : "));  Serial.print(configuration.OPTION.transmissionPower, BIN);Serial.print(" -> "); Serial.println(configuration.OPTION.getTransmissionPowerDescription());

	Serial.println("----------------------------------------");

}

```

And please recheck your connection schema with the schema on the article not on Github library, and check that you have the last version of the library, I fix sometings 2 days ago.

Bye Renzo

fransolomon, 2019-12-25:
Hi Renzo;

Hi.
I have done the following things:
a) I have tested your program as is but using my configuration pins. In my case for my arduino mega:
SoftwareSerial mySerial(10, 11); 
LoRa_E32 e32ttl100(&mySerial,12, 14, 13);
In the program it freezes in this line:
"ResponseStatus rs = e32ttl100.sendMessage (" Hello, world? "); // OK The message is received on the other device"
If I comment the line the program continues and does not freeze at this point. But I still have some problems.
b) I have reviewed all the placement of the components and even the M0 and M1 pins I have put them directly to ground. This does not seem to have changed much.
c) I have also isolated the module power supplies, feeding this apart to a good 5 volt power supply and joining all the masses.
d) I changed the pins before using (& mySerial, 5, 7, 6); and now I use (& mySerial, 12, 14, 13). You know if the M0, M1 and AUX pins used by your library require something special. Have interruptions for example.

I begin to think that the problem comes through the mega arduino itself, in a couple of weeks I will have a pair of arduino UNO and try to test your library with them.

Another detail that has caught my attention. I have always had to use your library to work well, serial ports by SOFTWARE. The arduino Mega has 4 serial ports per hardware and when I configure them directly I cannot send messages. Maybe I'm doing it wrong ?:
LoRa_E32 e32ttl100 (17,16,12, 14, 13);
void setup () {
   Serial2.begin (9600);
}

a.1) Once the line mentioned by the program has been commented, send the message "Hello, world?" and is received by the other LORA module (module B). Then from module B I start sending another message "for example 12345". Then module A receives the message but when I change the message to a shorter one, the program freezes.

best regards Fran

xreef, 2019-12-27:
I tested the example with a1 case but work for me and my Arduino UNO, I'm going to try with Mega.
Thanks for your patience Renzo

xreef, 2019-12-27:
I'm doing some test with mega and there is a problem, I'm going to discover in the next week.
Bye Renzo

xreef, 2019-12-27:
I check with my Arduino Mega, and my error is that I don't use a correct RX pin as Arduino guide describe.

https://www.arduino.cc/en/Reference/softwareSerial

Not all pins on the Mega and Mega 2560 support change interrupts, so only the following can be used for RX: 10, 11, 12, 13, 14, 15, 50, 51, 52, 53, A8 (62), A9 (63), A10 (64), A11 (65), A12 (66), A13 (67), A14 (68), A15 (69).

Now I can send without problem and receive without problem on my Mega without any freeze.

I use this configuration
SoftwareSerial mySerial(11, 5);
LoRa_E32 e32ttl100(&mySerial, 3, 7, 6);

Now work correctly.

Please update the libray, uncomment 
#define LoRa_E32_DEBUG

[image]

And send me your log, I try to understand.

Bye Renzo

fransolomon, 2019-12-28:
Sorry, I owe you an apology. A few days ago you asked me: "and check that you have the last version of the library, I fix sometings 2 days ago." This morning have become aware that i didn't have the updated library. I had two ZIP files with your git repository and I confused the versions. The old library was installed all the time and not the new one. This morning I noticed and updated the library. And it's going great, I've compiled all my arduino sketch and your examples and I'm really sorry for the mistake I made. Everything is correct, if later I can help you with tests I will be happy to help you.

xreef, 2019-12-28:
Thanks for your feedback, I'm happy that you have resolved your issue, in the next time I publish my shield, but I already worked on Arduino Mega one and surely I need to test It. If you like I need help on my site too.
Keep in touch Renzo

closed 2019-12-28 by xreef