xreef/LoRa_E32_Series_Library #3: Timeout Error!

muhammedoney, 2019-12-26:
I had timeout error. How can i fix it?

xreef, 2019-12-26:
You must send me more detail.
It can be caused by
Wrong connection, check Ir
Aux is not pulled and my timeout is not sufficient up or not connected, try connect
Damaged or not good powered up device

But if you send me some information I can try to help.

Bye Renzo

muhammedoney, 2019-12-26:
im using arduino mega
for uart usage i used 19th(rx1) and 18th(tx1) pins on Mega. attached tx of E32 to 19th pin and rx to 18th pin. After that i organized the code like that :  LoRa_E32 e32ttl(18, 19, 5, 7, 6);
my schematic connections are same as in this link : https://create.arduino.cc/projecthub/xreef/lora-e32-device-for-arduino-esp32-or-esp8266-library-728a86
when i open serial port i see these:
Init AUX pin!
Init M0 pin!
Init M1 pin!
Begin ex
Begin Software Serial
Begin 
MODE NORMAL!
Timeout error!
Hi, I'm going to send message!
Send message: Hello, world? size: 13
Timeout error!
Timeout!!
Send message: hey
 size: 4
Timeout error!

muhammedoney, 2019-12-26:
i checked the voltage on the E32. That was 3.3V and corrected it. Timeout has dissappeard but there is not data communication with the other radio
i cannot send data with transparent comm script

xreef, 2019-12-26:
Hi, please refer to my site mischianti.org for the schema, they are surely updated.
The supply voltage must be 5v, I correct all schema on my site.
Bye Renzo

xreef, 2019-12-27:
Try to set configuration, sometime the device is not correctly configured, try this with this configuration:

```
	ResponseStructContainer c;
	c = e32ttl100.getConfiguration();
	Configuration configuration = *(Configuration*) c.data;
	configuration.ADDL = 3;
	configuration.ADDH = 0;
	configuration.CHAN = 0x04;
	configuration.OPTION.fixedTransmission = FT_TRANSPARENT_TRANSMISSION;
	configuration.OPTION.wirelessWakeupTime = WAKE_UP_250;

    configuration.OPTION.fec = FEC_1_ON;
    configuration.OPTION.ioDriveMode = IO_D_MODE_PUSH_PULLS_PULL_UPS;
    configuration.OPTION.transmissionPower = POWER_20;

    configuration.SPED.airDataRate = AIR_DATA_RATE_010_24;
    configuration.SPED.uartBaudRate = UART_BPS_9600;
    configuration.SPED.uartParity = MODE_00_8N1;

    e32ttl100.setConfiguration(configuration, WRITE_CFG_PWR_DWN_SAVE);
```

daochoi, 2019-12-27:
Tôi cần giảm thời gian khi lam bài thi trên web.đó cái cái timeout ,tôi ko rành về code nhưng tôi đã làm dc 1 lần và giờ ko tìm lại dc cái đoạn code đó,làm ơn chỉ dùm ạ

xreef, 2019-12-27:
I think that there is a problem with arduino mega check issue #2 . I'm going to discover in the next week.
Bye Renzo

xreef, 2019-12-27:
I discover the problem, as you can see on Arduino guide you can use only some pins of Mega as receiving (RX) of software serial

https://www.arduino.cc/en/Reference/softwareSerial

Not all pins on the Mega and Mega 2560 support change interrupts, so only the following can be used for RX: 10, 11, 12, 13, 14, 15, 50, 51, 52, 53, A8 (62), A9 (63), A10 (64), A11 (65), A12 (66), A13 (67), A14 (68), A15 (69).

So you must use one of that pins, 18 is not correct pins.

Please send me a feedback thanks Renzo

muhammedoney, 2019-12-30:
I added the setting configuration section you typed, I used to communicate using two arduino nano, but now I can't do it either.
Now when I try to communicate, I get the message:
Init AUX pin!
Init M0 pin!
Init M1 pin!
Begin ex
Begin Software Serial
Begin
MODE NORMAL!
AUX HIGH!
Complete!
MODE PROGRAM / SLEEP!
AUX HIGH!
Complete!
3
MODE PROGRAM / SLEEP!
AUX HIGH!
Complete!
3
AUX HIGH!
Complete!
Clear buffer ... ok!
----------------------------------------
HEAD BIN INSIDE: 11000000 192 C0
----------------------------------------
MODE NORMAL!
AUX HIGH!
Complete!
Hi, I'm going to send a message!
Send message: Hello, world? size: 13
AUX HIGH!
Complete!
Clear buffer ... ok!
Success
Send message: hey
 size: 4
AUX HIGH!
Complete!
Clear buffer ... ok!

everything looks normal, but the message I wrote to the serial screen is not forwarded.
I think I burned one of my radios. So I can't experiment with mega. I will notify you again when I receive a new radio
thanks for your updates

muhammedoney, 2019-12-30:
I added the configuration section you typed, I used to communicate using two arduino nano, but now I can't do it either.
Now when I try to communicate, I get the message:
Init AUX pin!
Init M0 pin!
Init M1 pin!
Begin ex
Begin Software Serial
Begin 
MODE NORMAL!
AUX HIGH!
Complete!
MODE PROGRAM/SLEEP!
AUX HIGH!
Complete!
3
MODE PROGRAM/SLEEP!
AUX HIGH!
Complete!
3
AUX HIGH!
Complete!
Clear buffer...ok!

HEAD BIN INSIDE: 11000000 192 C0

MODE NORMAL!
AUX HIGH!
Complete!
Hi, I'm going to send message!
Send message: Hello, world? size: 13
AUX HIGH!
Complete!
Clear buffer...ok!
Success
Send message: hey
 size: 4
AUX HIGH!
Complete!
Clear buffer...ok!
everything looks normal, but the message I wrote to the serial screen is not forwarded.
I think I burned one of my radios. So I can't experiment with mega. I will notify you again when I receive a new radio
thanks for your updates

xreef, 2020-02-06:
I retested and seems working good. Thanks

closed 2020-02-06 by xreef