xreef/LoRa_E32_Series_Library #4: exemple not working with ESP32

pauldvg, 2020-01-19:
Hello,
First of all thank you for your work, I used your library successfully with an Arduino board.
Now I want to use it with an ESP32, but I can't get even the simplest exemple to work because of a problem with SoftwareSerial which does not exist for ESP32, even thought I'm trying to use the hardware serial port (GPIO 1 and 3)..
How did you manage to make it work with an ESP32? Can you provide a exemple code that works for the ESP32 ?
Thank you very much

pauldvg, 2020-01-20:
I finally managed to get it working by removing all reference to the SoftwareSerial class and objets in your library. I simply used the Serial object with `LoRa_E32 e32ttl(&Serial1, LoRa_AUX, LoRa_M0, LoRa_M1); // serial port, aux, m0, m1, baudrate`

Thank you

xreef, 2020-01-20:
Hi paul, 
Sorry if I write only now.
I'm going tuo fix the problem as soon as possible.
Thanks tuo open this issue Renzo

xreef, 2020-01-20:
Hi pauldvg,
I add a check on esp32 to disable by default SoftwareSerial.

But If you remove the check #ifndef ESP32 you can enable SoftwareSerial with esp32 
and work with espsoftwareserial library without problem 
https://github.com/plerup/espsoftwareserial

With that library you can use softwareserial with esp32.

Bye and thanks Renzo

pauldvg, 2020-01-23:
Thank you for your very quick reaction ! I will try the new version

xreef, 2020-02-06:
Tested and working. Thanks

closed 2020-02-06 by xreef