xreef/LoRa_E32_Series_Library #5: Any possibility to get RSSI or something compareable for the E32868T30D?

knopserl, 2020-01-26:
Are there any possibility to get RSSI value, something compareable or any means to monitor signal/communication quality/strength for the E32868T30D?
I guess the module itself does not provide that? Does it make sense to contact EBYTE to enhance their firmware for the module to implement that?
I have found statements and code that indicates, that Semtech SX1276/78 supports RSSI (which is most likely used by the E32 modules)

xreef, 2020-01-27:
Hi, 
in the datasheet they write that RSSI is not support, but in other EBYTE device It's present.
I think that It's a tecnical limitation, but this thevice is the cheapest.
If you want ask to EBYTE fell free to use this channel as reference.
Thanks Bye RM

knopserl, 2020-02-04:
@xreef I rather tend to think, that is not a limitation of the LoRa chip as all LoRa chips (from Semtech or other, support RSSI), in fact I have not seen any RF chip which does not support it.
I think they forgot to forward it in the serial communication interface they put in between at this modules. But of course I can contact EBYTE and ask, but it makes all the sense or it makes no sense if they dont support it.

xreef, 2020-02-06:
I try to contact EBYTE more than one time, but the response is very fast but I  was never able to get in touch with the technicians only with the first assistance.
I also tried to tell him that if they sent me other devices besides the e32s that I had, I would have made them free of charge, but I did not even receive a response.
I think that they are not interested to Arduino world.
Bye Renzo

xreef, 2020-02-07:
very similar to e32 exist a UART variant of e22, but package is SMD
E22-900T30S
E22-400T30S
E22-900T22S
E22-400T22S
and this library not cover this device.
I ask to EBYTE to give me 2 e22 to create a library like e32, It’s a simple work with e32 library like base (in 2 week is ready), but I never receive response.

Bye Renzo

knopserl, 2020-02-07:
@xreef I contacted EBYTE and here is the answer:

Dear Heinz, 

Thanks for choosing Ebyte. Sorry for the inconvenience. 

The E32 and E28 series don't have RSSI function because of hardware limitation. 
The E22 series support the RSSI function, please view this page to see various models of E22: http://www.ebyte.com/en/product-class.aspx?key=E22 

Please feel free to ask other questions about our products and i can also offer you discounted prices for your large orders. 
---------------------------------Best Regards,Sales Engineer:Robin HuEmail: sales06@ebyte.comPhone: +86-15817307997(whatsapp)Website: http://www.ebyte.com/en/Aliexpress：https://www.aliexpress.com/store/4836041Address:Innovation Center D347,4# XI-XIN road,High-tech district(west),Chengdu,Sichuan,China

xreef, 2020-02-07:
The adaptation of the library would be a trivial matter, but I would have expected a minimum of collaboration and support from Ebyte.
Without devices tuo test Its difficult to develop.
Bye Renzo

xreef, 2020-03-06:
Finally I order 2 e22, stay connected for new library version.
Bye Renzo

closed 2020-03-06 by xreef

knopserl, 2020-03-22:
@xreef I did the same :-)