xreef/LoRa_E32_Series_Library #6: ESP32 HardwareSerial custom pins

kind3r, 2020-02-06:
I can't for the love of C++ (read hate) set custom pins with HardwareSerial. Re-defining `RX1` and `TX1` for `Serial1` does not work, only way to set other pins seems to be via `HardwareSerial.begin()` which is called in `LoRa_E32::begin()` via `NeedsStream::begin()`.

Any ideea how I can work around this ?

I can create a new constructor to specify the pins and try to create a pull request.

xreef, 2020-02-06:
Hi Kind3r,
Thank you for notifying me of this shortcoming and for the code,
I can't grab directly your pull request because compatibility problem steal some precious code from you pull request, and I create a new branch here
https://github.com/xreef/LoRa_E32_Series_Library/tree/Add_rx_tx_pins_for_esp32_device
I hope It's work for you, please give me a feedback.

If you give me an OK about this code, when I have sometime I test backward compatibility and I put all in a master branch.

Thanks again Renzo

kind3r, 2020-02-07:
Ok, I can confirm it's working. Thanks !

xreef, 2020-02-07:
Ok thanks for your feedback, I'm going to test on other device than I put all in a master branch.
Thanks again bye Renzo

closed 2020-02-07 by xreef