xreef/LoRa_E32_Series_Library #7: Serial mode during MODE_3_SLEEP

kind3r, 2020-02-06:
As per the docs 9600 8N1 is required in mode 3 for programming, regardles of the normal operation communication settings.
I put this here for future reference, not that you would really need to change the baud rate in normal mode unless you want to save a tiny bit of power while sending your data to the module.

xreef, 2020-02-06:
Thanks for your information, 
do you think that It's a good idea to check if there is a correct Serial baudrate and config and return a specified error when mode 3 is activated?

Thanks again Renzo

kind3r, 2020-02-06:
As a quick fix yes, so others don't run into issue. For me this is not a problem.

xreef, 2020-03-06:
I add the control.
Thanks again bye Renzo

closed 2020-03-06 by xreef