xreef/LoRa_E32_Series_Library #10: Re-transmitting problem

mrekin, 2020-02-26:
Hi!
I'm not familiar with esp8266 and this might be not a problem.
I'm trying to realize such code (just to test modules and communication distance)

  _```
// If something available
  if (e32ttl100.available() > 1) {
    // read the String message
    ResponseContainer rc = e32ttl100.receiveMessage();
    // Is something goes wrong print error
    if (rc.status.code != 1) {
      Serial.println(rc.status.getResponseDescription());
    } else {
      // Print the data received
      Serial.println("Recieved: ");
      Serial.println(rc.data);
      ResponseStatus rs = e32ttl100.sendMessage(strcat("Resending: ", rc.data.c_str()));

    }
  }
```_

and some thing wrong with rc.data - it contains old messages (looks like some buffer not cleans after reading).
Is it my code problem or something else ?
Wemos D1 mini + E32-433T20D, normal mode (M0 and M1 set to GND)

xreef, 2020-02-28:
Hi,
sorry if I response you only now

the usage of strcat create issue, you are going to add on a memory destination without remove It.
Try this

	// If something available
	if (e32ttl100.available() > 1) {
		// read the String message
		ResponseContainer rc = e32ttl100.receiveMessage();
		// Is something goes wrong print error
		if (rc.status.code != 1) {
			Serial.println(rc.status.getResponseDescription());
		} else {
			// Print the data received
			Serial.println("Recieved: ");
			Serial.println(rc.data);
			ResponseStatus rs = e32ttl100.sendMessage("Receiving: " + rc.data);

		}
	}

Give me a feedback.
Bye Renzo

closed 2020-02-28 by xreef

mrekin, 2020-03-02:
Thanks for reply with right direction).
Problem was in "sender" device with input buffer.