xreef/LoRa_E32_Series_Library #11: how to send and recieve custom messages?

halukmy, 2020-03-04:
https://github.com/sandeepmistry/arduino-LoRa/blob/master/examples/LoRaDuplex/LoRaDuplex.ino

can you make simplier like this repo?

xreef, 2020-03-04:
Hi,
you can check the tutorial here 
https://www.mischianti.org/it/2019/10/15/lora-e32-per-arduino-esp32-o-esp8266-specifiche-ed-utilizzo-base-part-1/

And here the example
https://github.com/xreef/LoRa_E32_Series_Library/blob/master/examples/sendReceiveTransparentTransmissionMessage/sendReceiveTransparentTransmissionMessage.ino

xreef, 2020-03-04:
Hi
I'm going to do a sample similar.
Bye Renzo

halukmy, 2020-03-05:
@xreef  thanks for inform waiiting updates

hope its comes soon make it simple as arduino-lora please

xreef, 2020-03-05:
I think It's ok but untested

```
#include "Arduino.h"
#include "LoRa_E32.h"

LoRa_E32 e32ttl100(4, 5, 3, 7, 6);

void setup() {
	Serial.begin(9600);
	delay(500);

	e32ttl100.begin();

}
long lastSendTime = 0;        // last send time
int interval = 3000;          // interval between sends

void loop() {
	  if (millis() - lastSendTime > interval) {
	    String message = "HeLoRa World!";   // send a message
		ResponseStatus rs = e32ttl100.sendMessage(message);
		
	    Serial.println("Sending " + message);
		Serial.print("Sending state: ");
		Serial.println(rs.getResponseDescription());		// Check If there is some problem of succesfully send
		
	    lastSendTime = millis();            // timestamp the message
	    interval = random(2000) + 2000;    // 3-4 seconds
	  }

	if (e32ttl100.available() > 1) {
		Serial.println("----------");
		ResponseContainer rc = e32ttl100.receiveMessage();

		if (rc.status.code != 1) {
			rc.status.getResponseDescription(); // if something goes wrong
		} else {
			Serial.println(rc.data); // if It's all ok
		}
	}
}
```

halukmy, 2020-03-06:
i just oredered devices x4 on the way, i wonder is it have control about spread spectrum or coding rate or etc can we config ?

xreef, 2020-03-06:
It's quite simple to config, you have all examples and parameters in this article
https://www.mischianti.org/2019/10/29/lora-e32-device-for-arduino-esp32-or-esp8266-configuration-part-3/
Bye Renzo

closed 2020-03-07 by xreef