xreef/LoRa_E32_Series_Library #12: E32 to LoRaWan

darkjwl, 2020-03-16:
Can I connect e32 433 MHz module to LoRa WAN? If can then how could I connect it ? I have already tried to connect with LoRaWAN but I can’t ! If anyone done this please share your experience 

Thanks

xreef, 2020-03-16:
Hi darkjwl,
I think that It isn't possible, but I keep this post open if other have a different answer.
Bye Renzo

darkjwl, 2020-03-17:
Thank you @ xreef 
Which LoRa Module is best for LoRaWAN communication. I also have RFM95 but i found it's covered short range (800m) only. do u have any suggestion ?

xreef, 2020-03-17:
I know e78 from same vendor of e32, but like e32 no more tha datasheet is present, so i can hink you must search other device. Sorry.
Bye Renzo

closed 2020-03-20 by xreef