xreef/LoRa_E32_Series_Library #14: Parameter Setting and ''No response from device!'' Error

cglkr97, 2020-04-12:
Hey Renzo!

I upload your setConfiguration code to my E32 device and set my parameters successfully as you can see in the picture below. 

[image]

But right after it, when I upload the getConfiguration code, the parameters seem to get changed even if I didn't change any parameters as you can see below. I also get the ''No response from device'' error when I upload the .

[image]

This, of course, causes the module to fail when sending data because I keep sending the data to the wrong address I guess. Here are my wirings down below. I made a little Arduino Uno on a breadboard that works on 8MHz crystal with 3.3V. I program it using an Arduino Uno board which works fine. I supply 3.3V to the breadboard with a voltage regulator which also seems to be working fine. 

[image]

What do you think is causing this? What do you think I'm doing wrong here? 

Thanks in advance! And thank you for the library you created! :)

xreef, 2020-04-12:
Hi,
in the first screen there is a response "No response from device", I think that can be problem with power supply, 

but my fist advise is to attach AUX pin that Its very usefully to have a correct error response.

What kinf of regulator you use?
Bye Renzo

cglkr97, 2020-04-12:
It's an MCP1702 fixed 3.3V and as we speak I'm measuring 3.26V across the module. 
> but my fist advise is to attach AUX pin that Its very usefully to have a correct error response.
> 
So should I connect AUX to any digital pin right? A 4.7k pull-up resistor perhaps? 

Thank you for your lightning fast response!

xreef, 2020-04-12:
250mha I think It's sufficient.
Yes 4.7k pull-up resistor,
and If you need more information decomment 
// #define LoRa_E32_DEBUG
and check debug serial
#define DEBUG_PRINTER Serial1

to get verbose information.

Bye Renzo

cglkr97, 2020-04-13:
Hey, again Renzo!

> // #define LoRa_E32_DEBUG
and check debug serial
#define DEBUG_PRINTER Serial1
> 

As you told me I decommented the first line and made sure the second line is also decommented. I  connected AUX to D8 along with a 4.7k pull-up resistor. But I still received the same error with no further info on why the error happens. 

So this morning I got my hands into the EBYTE parameter setting software. I connected my module to my computer via the FTDI module and it seems that I set the parameters successfully by using the software (see pic below). But when I come back to your library and upload the getConfiguration code I still get the same error and the parameter values do not match the ones I set earlier with the EBYTE software.

[image]

At this point, I started to think my wiring on my Arduino on Breadboard(which is in the above pic) is wrong. Yet, I check my E32 wirings repeatedly and they seem okay. The regulator is fine, I measure 3.26V from the power line. I also changed my Atmega328P chip. And still got the same error along with the wrong parameter values. 

But the bigger problem here is, I guess because of this parameter setting problem my modules can't make the fixed transmission. I upload your codes in the sendRecieveFixedTransmissionPoint folder. I'm sure that I type the correct receiver address and channel to the sendFixedTransmisson code but the sender sends the message yet the receiver doesn't receive it. There's only ''Start listening!'' text on the serial monitor of the receiver. 

Sorry to be a pain but I'm stuck here. I would appreciate and welcome any idea on why this happens!
Thanks!

cglkr97, 2020-04-13:
I think I spot the problem. The function **e32ttl100.available** always returns 0 in my setup. And because of this the following code never gets executed thus I can't read any messages from the receiving device.

`if (e32ttl100.available()>1) {
	  // read the String message
	ResponseContainer rc = e32ttl100.receiveMessage();`
 
But what might be causing this?

xreef, 2020-04-13:
Hi,
The second line is decommented but Serial1 Is set, probably you must set Serial not Serial1 (the serial you use for debug monitor).

Now retry to getConfiguration with a bigger pull up resistor (10k 20k or try to remove) from rx tx.

Than write me the result. Thanks Renzo

cglkr97, 2020-04-17:
Hey Renzo!

Sorry for the late response. Yes, I decommented both of the lines as you can see. But I search for **''Serial1''** in LoRa_E32.h and LoRa_E32.cpp and the searcher finds nothing. So I couldn't change it which is what I supposed to do anyways I think.

[image]

As you said, I added the 10k pull-up resistors to my TX and RX and getConfiguration still returned the same error.  I also tried 20k and 50k resistors but there was no change. 
Anything else I should try?
Thanks Renzo!

xreef, 2020-04-18:
Please send me the complete Serial monitor output.
When decomment DEBUG line you must receive a lot of information, if don't you not decomment correct line.
Bye Renzo

cglkr97, 2020-04-24:
Hey Renzo!

Sorry for the very late response. I've been battling with Covid-19 and proudly can say that I beat it!
And I also solved the problem. I just realized that I messed up the Rx and Tx pins. The truth is I got confused by the code. Because in the code you wrote 

> * RX         ----- PIN 2 (PullUP & Voltage divider)
> * TX         ----- PIN 3 (PullUP)   
And then went on declaring

> LoRa_E32 e32ttl100(2, 3); // RX, TX

So I naturally went on connecting the Rx pin of the E32 to Pin 2 of my stand-alone Arduino and then declare Pin 2 as the Rx in the constructor. This was not correct because I knew that you're supposed to connect Rxs and Txs crosswise in a two wire communication system. I actually did question why you declared Pin2 as Rx and Pin3 as Tx but I guess I preferred to trust the code. 

So this was a misunderstanding obviously. But is it only me? Has anybody else contacted you about a similar kind of misunderstanding or an issue similar to mine?
Thank you very much tho. I'm using E32 modules for my senior project and your library is exactly what I needed. You wrote a great one. Far superior than anything else on GitHub with many great examples. 
Best, 
C.

xreef, 2020-04-26:
Ahhh.. ok.. 
I think I must check that misunderstanding, If you have some time to spent do a pull request with correction.
Thanks for your support, when I find some time I fix It.
Bye Renzo

closed 2020-05-18 by xreef