xreef/LoRa_E32_Series_Library #17: sendFixedMessage Parameters Issue

cglkr97, 2020-06-01:
Ciao Renzo!

I'm trying to do a project where I tell my sender E32 which receiver address it will send the message to. I give the address info via serial monitor.  I ask the user to provide the ADDH, ADDL, CHAN bytes. 
For example I give the ADDH=0 ADDL=0 and CHAN=17 and pass these into _sendFixedMessage_ as parameters.

` ResponseStatus wakeMes = e32ttl.sendFixedMessage(ADDH,ADDL,CHAN, "Sensor Data");`

But when the sender sends to this address, the receiver gives overflows and data get messed up since I can't write the CHAN byte as _0x17_. 

I've tried writing it as,
` ResponseStatus wakeMes = e32ttl.sendFixedMessage(ADDH,ADDL,Serial.print(23,HEX), "Sensor Data");`
but it again gave overflow and data got messed up.

Is there any other way that I can store the serial byte input as hex and then pass it into  _sendFixedMessage_ function as address parameter?
 
I've tried doing this,
` byte b =Serial.println(a,HEX);
  Serial.println(b);`
but again, it always turned something incorrect.

Or is there any other way that I could play with your library to make the _sendFixedMessage_ function to accept the user input byte without writing it as _0xCHAN_?

Thanks,
C.

xreef, 2020-06-01:
Hi cglkr97,
can you send the complete code PLZ.
Thanks Renzo

cglkr97, 2020-06-02:
Hi Renzo,
Here is my code.
[gatewayFixedMessage2.txt](https://github.com/xreef/LoRa_E32_Series_Library/files/4714184/gatewayFixedMessage2.txt)

Thanks,

xreef, 2020-06-06:
Hi cglkr97,
I moved this discussion on my forum because become quite interesting.
https://www.mischianti.org/forums/topic/ebyte-e32-sendfixedmessage-parameters-issue/
Check It please.
Bye Renzo

closed 2020-06-26 by xreef