xreef/LoRa_E32_Series_Library #20: return 0 value problem

aliustunelin, 2020-08-29:
I'm working with arduino nano and also I tried uno. There is no problem in getting the value, but after a while, arduino nano: 26th value, value is uno: 40th value. nano always return 0.
What could be the reason for this?

xreef, 2020-08-29:
Hi aliustunelin,
It's very strange, can you send me the code and wiring? so I can replicate the situation.
If you use the site forum you can attach the files.
Bye Renzo

aliustunelin, 2020-08-29:
It is solved, the connection diagram is different from the ones in your comment. I changed the m0 and m1 pins.

So where can I find the code that I can connect to peer to peer
so I need to transmit GPS data from one device to another rather than configuring it

xreef, 2020-08-29:
You can find informtion about point to point transmission here https://www.mischianti.org/2019/11/10/lora-e32-device-for-arduino-esp32-or-esp8266-fixed-transmission-part-4/ , but If you want manage a sleep mode of your remote device you must refer here https://www.mischianti.org/2019/12/28/lora-e32-device-for-arduino-esp32-or-esp8266-wor-wake-on-radio-the-microcontroller-also-and-new-arduino-shield-part-6/ .

If you explain your project I can give you more detailed information.

Bye Renzo

aliustunelin, 2020-08-29:
In fact, the project is to take GPS data in an area of ​​1200 meters. Includes payload, GPS sensor and Arduino Nano.
The uno in the center will get the GPS data in the nano.
I need to setup this communication with Lora ebyte e32 433mhz

xreef, 2020-08-29:
Hi,
you can follow the tutoria of fixed trasmission, or use the monitor configuration con server tuo receive all messages from other client..
Bye Renzo

aliustunelin, 2020-08-31:
Hi,
do you have the wiring diagram for arduino mega 2560 and ebyte e32?

xreef, 2020-08-31:
Hi,
no sorry, but other people had try to connect Mega, and you can use the same schema of Arduino but you must pay attention to the SoftwareSerial RX pin of Mega, you can find more info here
https://www.mischianti.org/forums/topic/mega-2560-e32-ebyte-frozen-in-example-sendreceivetransparenttransmissionmessage/

aliustunelin, 2020-08-31:
So if we wanted to use hardwareserial pins, what would the pin order be like?
For example
SoftwareSerial mySerial (14, 15);
LoRa_E32 e32ttl100 (& mySerial, 3, 7, 6);

in this way

xreef, 2020-08-31:
For hardware serial you must use constructor like

```cpp
LoRa_E32 e32ttl1w(&Serial2, AUX, M0, M1);
```

https://www.mischianti.org/forums/topic/ebyte-lora-e32-not-working-on-esp32-sends-a-random-number-the-other-reads/

Bye Renzo

aliustunelin, 2020-09-02:
how can i tell if lora ebyte e32 433mhz module is burned?

xreef, 2020-09-02:
I don't understand very well the last  sentence..
 but normally to check if device Is burned I use my shield, first I try to read the configuration, than I try to write configuration.

But It's huppened that the device start to have trouble, than I'm going to reset module
`ResponseStatus LoRa_E32::resetModule()`
than repeat the upper operation.

But a good solution Is to use a board with EByte software to be sure of that.

Bye Renzo

patricioaranguiz, 2020-09-06:
Hi Renzo, I have a similar problem, I'll tell you ...

I have a Node MCU which connects to AWS to insert the data to Dynamodb, the Node MCU has a cron that runs every 10 minutes where it sends a message to request information from an Arduino UNO, but after a few hours it stops function and they cannot communicate.

Should I reset the module that is in the Arduino UNO? or what do you recommend?

Thanks

xreef, 2020-09-07:
Hi patricioaranguiz,
I can't find the issue, I need to debug the code, 
but If you open a Topic on my forum https://www.mischianti.org/forums/forum/yours-project/ and attach the files I try do replicate and test the code to find the issue.
Bye Renzo

closed 2020-09-11 by xreef

bratello, 2022-03-15:
Hi Renzo. I have the similar problem on Raspberry. I use the nodejs or python for e32 communication on Raspberry, and your library on ESP32 as the target device. So I'm truing to implement the same API on Raspi through the script. The Raspi establishes the 0x17 channel (Service Channel) and setup the target device through the broadcast request - the target device starting listen on 0x4 channel (Communication Channel) and answers to the set of requests from Raspi. After 10 or more requests, the Raspi code stuck - same on python and nodejs. The one difference - on nodejs I got some garbage immediately after send. I'm looking to your API and trying to understand what's missing in my script. As I understand LoRa_E32::waitCompleteResponse checking the AUX state for IO completion identification - it may be useful on send, but on read I see the AUX state HIGH but the data still not arrived to the script. But my main question - LoRa_E32::cleanUARTBuffer. Why you need this cleanup after IO operation? Maybe you also experienced the similar issue when you developed your API, and you can point me some important thing. 
Thanks