xreef/LoRa_E32_Series_Library #21: gestione modalità m0 e m1 non ottimizzata

AB-informatica-service, 2020-09-19:
salve lasciando gestire m0 e m1 alla libreria dopo alcuni messaggi non viene più impostato m1 correttamente  
esempio:  dopo circa 10 messaggi quando viene inviato un messaggio il modem risveglia il processore ma m1 rimane a 3.3v e non è possibile ricevere e inviare il messaggio. impostando m0 e m1 in modalità zero non si presenta mai questo problema 
dettagli: questo problema si presenta assolutamente a caso non ha uno schema preciso a volte succede dopo 4 messaggi a volte succede dopo 10 messaggi quando m1 non si porta a 0 può succedere anche più di 1 volta di fila e in questo caso il messaggio non viene trasmesso al processore. per ulteriori dettagli sono disponibile grazie

xreef, 2020-09-21:
Ciao,
prova ad inserire delle resistenze per aiutare la messa a terra del dispositivo, quando alimentato a 5v ha un'intensita che non sempre è facile da far cambiare stato con i pin del microcontrollore.
Fai un ulteriore test, scollega l'E32 e controlla che il microcontrollore sia correttamente a LOW.
Ciao Renzo

AB-informatica-service, 2020-09-22:
Ciao Renzo, tu mi dici di vedere se senza l'E32 la libreria lavora bene giusto?Mentre per le resistenze intendi mettere tipo delle resistenze da 180 ohm sul tra GND e M1?

AB-informatica-service, 2020-09-22:
Ho fatto dei test a 5V: ho fissato su breadboard MO e M1 a GND e con un multimetro ho misurato l'uscita del MCU ed effettivamente funziona sempre bene, cambia stato come dovrebbe.
Quindi a questo punto la libreria non dovrebbe avere problemi ma è il modem che non va a GND però mi chiedo come può essere possibili dato che Atmega328P quando setti digitalWrite(pin, LOW) va a GND diretta, come fa i modem a rimanere sui 3.8V e non andare a GND?

Grazie

xreef, 2020-09-23:
E' quello che ti dicevo, ha di suo un'uscita di corrente, per poterlo mandare a GND prova ad inserire una resistenza pull-down (collegata al GND) per aiutare il controllore a mandarlo a terra, sulla versione 1W che usi probabilmente ha un amperaggio elevato e fatica a controbattere.
Fammi sapere.
Ciao Renzo

AB-informatica-service, 2020-09-23:
Ho fatto una giornata di test, ho provato con diverse PULL-DOWN (10k, 4.7k, 1k, 560) ma il risultato non cambia, molte volte non setta M1 a LOW e si perde il messaggio.
Ho provato a utilizzare un pin digitale, il 13 per l'esattezza, e cambiare in maniera manuale, con digitalWrite, il pin e vedere se cosi andava meglio, allora ho avuto degli ottimo risultati ovvero su 20 messaggi 2/3 li perde mentre lasciando gestire alla libreria su 20 ne perdo più della metà.

Ho pensato che magari l'MCU Atmega328Pnon riesce a contrastare la quantità di corrente che eroga il modem da M1 cosi ho fatto un test : ho comandato un piccolo relè a 5V con un contatto a scambio dove sul comune ho messo M1 del modem, sul normalmente aperto VCC e sul normalmente chiuso GND, ma anche qua alle volte perde messaggi, è proprio l'MCU che non mette a zero il pin (sia il 13, sia quello gestito dalla libreria).

Ora farò delle prove mettendo dei delay e scrivendo in seriale più messaggi per capire bene cosa sta succedendo durante il codice. 

Ti faccio sapere 

Grazie

xreef, 2020-09-23:
Prova a mettere un cavo diretto GND M0 M1, e metti la resistenza tra il pin del microcontrollore e il pin M0 M1 per vedere se riesce a variare.
Ciao RM

AB-informatica-service, 2020-09-25:
Ciao Renzo, ho messo a GND diretta M0 e M1 e ho messo un LED sul pin 7 gestito dalla tua libreria: non ne sbaglia una! Modifica sempre lo stato del LED e quindi la tua libreria non sembra avere problemi ma non capisco allora perchè collegando il modem alla volta non va LOW (o magari lo fa ma all'interno del modem chissà che succede). Fammi sapere se tu hai delle soluzioni. Grazie

AB-informatica-service, 2020-09-25:
Ho fatto ulteriori test e ho scoperto che mettendo M0 a GND fisso e gestire M1 dalla libreria va benissimo.

xreef, 2020-09-25:
M0 e M1 hanno un comune, e per mandarli a LOW gestiti dalla libreria puoi provare a metterli a GND e tra i pins M0  M1 e i pins del microcontrollore metti una resistenza così da permetterne il controllo.
Fammi sapere.
Ciao Renzo

AB-informatica-service, 2020-09-25:
Forse non ho capito, intendi che dovrei collegare a GND direttamente M0 e M1 e poi come li collego al MCU?

AB-informatica-service, 2020-09-26:
Rettifico: questa mattina non funziona neanche con GND a M0 fisso.
Ieri non perdeva nemmeno un messaggio.
Possibile che sia così instabile il modem?

AB-informatica-service, 2020-09-26:
Allora, ho notato che l'MCU non era stabile sulla Bread ora l'ho fissata e non ha più problemi. Con M0 a GND e M1 gestito dalla libreria non sbaglia un messaggio quindi OTTIMO!
Nel frattempo mi sono accorto che non ho collegato il pin di AUX al MCU ma me lo risveglia lostesso.... Com'è possibile?

xreef, 2020-09-27:
Per il collegamento ad M0 e M1 prova ad invertire 
[image]

Credo sia improbabile, probabilmente sei in sleep solo dell'E32 e non sei in sleep mode sull'ATMega.

Ciao RM

closed 2020-10-15 by xreef