xreef/LoRa_E32_Series_Library #30: Transparent transmission between 2 Wemos D1 lite + ES32-868T30D (1W) - message sent but no receive

Lucamenghini, 2021-07-25:
Hi Renzo. I have 2 Wemos D1 lite identical to the Wemos on your blog. I connected them exactly as you shown to one (one each) E32-868T30D. I started from your example "sendReceiveTransparentTransmissionE32_TTL_1W". One board is connected to a notebook and the other to another one notebook.
I upload to the first and to the second board the skecth below (little bit modified to remove the serialsoftware and use the harwdare on port).
The first board starts and I monitoring it by the serial monitor, it appear starting fine and looping for a signal. The second board is used as a transmitter with the same sketch. Reading the code I suppose pressing the reset button on the transmitter the board reboots and send a message: in fact the message "succesfull" is shown but on the serial monitor in the other board no message is shown.
Here you are the whole code. Sorry for my english, I'm italian as you but here I wrote in english to support the entire community.
Thank you for your support!
Best regards

/*

LoRa E32-TTL-1W
Send transparent transmission with 1W device
https://www.mischianti.org/2019/10/21/lora-e32-device-for-arduino-esp32-or-esp8266-library-part-2/
E32-TTL-100----- Arduino UNO or esp8266
M0 ----- 3.3v (To config) GND (To send) 7 (To dinamically manage)
M1 ----- 3.3v (To config) GND (To send) 6 (To dinamically manage)
TX ----- PIN 2 (PullUP)
RX ----- PIN 3 (PullUP & Voltage divider)
AUX ----- Not connected (5 if you connect)
VCC ----- 3.3v/5v
GND ----- GND
*/

#include "Arduino.h"
#define E32_TTL_1W
#define FREQUENCY_868
#include "LoRa_E32.h"

// SoftwareSerial mySerial(4, 5); // e32 TX e32 RX
// LoRa_E32 e32ttl100(&mySerial, 3, 7, 6);
LoRa_E32 e32ttl100(D2, D3); // e32 TX e32 RX

void printParameters(struct Configuration configuration) ;

void setup() {
Serial.begin(9600);
delay(500);

// Startup all pins and UART
e32ttl100.begin();

ResponseStructContainer c;
c = e32ttl100.getConfiguration();
Configuration configuration = *(Configuration*) c.data;
configuration.ADDL = 3;
configuration.ADDH = 0;
configuration.CHAN = 0x04;
configuration.OPTION.fixedTransmission = FT_TRANSPARENT_TRANSMISSION;
configuration.OPTION.wirelessWakeupTime = WAKE_UP_250;

configuration.OPTION.fec = FEC_1_ON;
configuration.OPTION.ioDriveMode = IO_D_MODE_PUSH_PULLS_PULL_UPS;
configuration.OPTION.transmissionPower = POWER_30;

configuration.SPED.airDataRate = AIR_DATA_RATE_010_24;
configuration.SPED.uartBaudRate = UART_BPS_9600;
configuration.SPED.uartParity = MODE_00_8N1;

e32ttl100.setConfiguration(configuration, WRITE_CFG_PWR_DWN_SAVE);
printParameters(configuration);
c.close();
Serial.println("Hi, I'm going to send message!");
// Send message
ResponseStatus rs = e32ttl100.sendMessage("Hello, world?"); // OK The message is received on the other device
// Check If there is some problem of succesfully send
Serial.println(rs.getResponseDescription());
}

void loop() {
// If something available
//Serial.println("continuity test point"); //This instruction is not executed unless you comment on the previous line "Serial.println(rs.getResponseDescription());"
if (e32ttl100.available() > 0) {
Serial.println("----------");
ResponseContainer rc = e32ttl100.receiveMessage();
Serial.println(rc.data);
// Is something goes wrong print error
if (rc.status.code != 1) {
rc.status.getResponseDescription();
} else {
// Print the data received
Serial.println(rc.data);
}
}
if (Serial.available()) {
String input = Serial.readString();
e32ttl100.sendMessage(input);
}
}
void printParameters(struct Configuration configuration) {
Serial.println("----------------------------------------");

Serial.print(F("HEAD : "));  Serial.print(configuration.HEAD, BIN);Serial.print(" ");Serial.print(configuration.HEAD, DEC);Serial.print(" ");Serial.println(configuration.HEAD, HEX);
Serial.println(F(" "));
Serial.print(F("AddH : "));  Serial.println(configuration.ADDH, DEC);
Serial.print(F("AddL : "));  Serial.println(configuration.ADDL, DEC);
Serial.print(F("Chan : "));  Serial.print(configuration.CHAN, DEC); Serial.print(" -> "); Serial.println(configuration.getChannelDescription());
Serial.println(F(" "));
Serial.print(F("SpeedParityBit     : "));  Serial.print(configuration.SPED.uartParity, BIN);Serial.print(" -> "); Serial.println(configuration.SPED.getUARTParityDescription());
Serial.print(F("SpeedUARTDatte  : "));  Serial.print(configuration.SPED.uartBaudRate, BIN);Serial.print(" -> "); Serial.println(configuration.SPED.getUARTBaudRate());
Serial.print(F("SpeedAirDataRate   : "));  Serial.print(configuration.SPED.airDataRate, BIN);Serial.print(" -> "); Serial.println(configuration.SPED.getAirDataRate());

Serial.print(F("OptionTrans        : "));  Serial.print(configuration.OPTION.fixedTransmission, BIN);Serial.print(" -> "); Serial.println(configuration.OPTION.getFixedTransmissionDescription());
Serial.print(F("OptionPullup       : "));  Serial.print(configuration.OPTION.ioDriveMode, BIN);Serial.print(" -> "); Serial.println(configuration.OPTION.getIODroveModeDescription());
Serial.print(F("OptionWakeup       : "));  Serial.print(configuration.OPTION.wirelessWakeupTime, BIN);Serial.print(" -> "); Serial.println(configuration.OPTION.getWirelessWakeUPTimeDescription());
Serial.print(F("OptionFEC          : "));  Serial.print(configuration.OPTION.fec, BIN);Serial.print(" -> "); Serial.println(configuration.OPTION.getFECDescription());
Serial.print(F("OptionPower        : "));  Serial.print(configuration.OPTION.transmissionPower, BIN);Serial.print(" -> "); Serial.println(configuration.OPTION.getTransmissionPowerDescription());

Serial.println("----------------------------------------");
}

xreef, 2021-07-26:
Hi Luca,
try to remove pull-up resistor on M0 and M1.
And verify that this pin go up and down as needed.
Bye Renzo

xreef, 2021-08-03:
Inactive

closed 2021-08-03 by xreef