xreef/LoRa_E32_Series_Library #33: I can't do two way communication

ErenKaymakci, 2021-08-23:
Hi, i am trying to two way comm with this codes:

#include "LoRa_E32.h"
#include <SoftwareSerial.h>
 
SoftwareSerial mySerial(10, 11);
 
/*
   Pins    Arduino Nano    Lora E32 433T20d
                  11                3
                  10                4
*/
 
LoRa_E32 e32ttl(&mySerial);
 
struct Signal {
  byte type=120;
  byte temp=10;
} data;
 
void setup() {
  Serial.begin(9600);
  e32ttl.begin();
  delay(500);
}
 
void loop() {
 
  ResponseStatus rs = e32ttl.sendFixedMessage(0, 5, 6, &data, sizeof(Signal));
  Serial.println(rs.getResponseDescription());
 
  delay(2000);
 
  while (e32ttl.available()  > 1) {
    ResponseStructContainer rsc = e32ttl.receiveMessage(sizeof(Signal));
    struct Signal data = *(Signal*) rsc.data;
    Serial.print("Type: ");
    Serial.println(data.type);
    Serial.print("Ambient temperature: ");
    //Serial.println(*(float*)(data.temp));
    rsc.close();
  }
}

#include "LoRa_E32.h"
#include <SoftwareSerial.h>
SoftwareSerial mySerial(10, 11); // Arduino RX <-- e32 TX, Arduino TX --> e32 RX
LoRa_E32 e32ttl(&mySerial);
 
struct Signal {
  byte type;
  byte temp;
} data;
 
void setup() {
  Serial.begin(9600);
  e32ttl.begin();
  delay(500);
}
 
void loop() {
  while (e32ttl.available()  > 1) {
 
    // Reading incoming message
    ResponseStructContainer rsc = e32ttl.receiveMessage(sizeof(Signal));
    struct Signal data = *(Signal*) rsc.data;
    Serial.print("incoming message: ");
    Serial.println(data.type);
    rsc.close();
 
    //Preparing packet data to be sent
    struct Signal  {
      byte type = 130;
      byte temp=90;
    } data2;
 
    //*(float*)(data2.temp) = 19.2;
 
 
    ResponseStatus rs = e32ttl.sendFixedMessage(0, 3, 6, &data2, sizeof(Signal));
    Serial.println(rs.getResponseDescription());
 
  }
 
}

My schema is like that:
[image]

Data is sending then i see "incoming message..." in serial port. But on the other port only "Success" message i can see. There is no "Type: ... " messages. How can i solve it ?

Thank you

xreef, 2021-09-02:
Hi Eren,
sorry i lost your issue, can you send me also the fixed transmission configuration of your devices.
Bye and sorry again Renzo

xreef, 2021-10-08:
No activity

closed 2021-10-08 by xreef