xreef/LoRa_E32_Series_Library #34: not reading the received data

RaspiGame, 2021-09-04:
> Hi, I have two of the LoRa E32 1W devices (https://es.aliexpress.com/item/32791508935.html?spm=a2g0s.9042311.0.0.274263c03mbIhS) with two arduino mega. I'm using this library and running the example named: sendReceiveTransparentTransmissionE32_TTL_1W.
> connecting all like specified:
>  * E32-TTL-100----- Arduino UNO or esp8266
>  * M0         ----- 3.3v (To config) GND (To send) 7 (To dinamically manage)
>  * M1         ----- 3.3v (To config) GND (To send) 6 (To dinamically manage)
>  * TX         ----- PIN 2 (PullUP)
>  * RX         ----- PIN 3 (PullUP & Voltage divider)
>  * AUX        ----- Not connected (5 if you connect)
>  * VCC        ----- 3.3v/5v
>  * GND        ----- GND

**the problem is when I send some data by SerialPort of arduinoTX, the SerialPort of arduinoRX don't receive nothing, despite that checking at the pin RX E32 with the oscilloscope, it show's that data is received.**

>  - I also added the following lines in *Loop* to skip the "possible library issue reading", but nothing different happened
> 
> ```
> if (mySerial.available() > 0){
>     Serial.println(mySerial.read());
> }
> ```

> - the other possible option was that arduino wasn't able to read the data from E32 due low voltage of arround 3v. So, I added this other lines to check it... And every time the message is received, the led flash.
>  ```
> if( digitalRead(4) == LOW ){
>   digitalWrite(13, HIGH);
>  }else{
>   digitalWrite(13, LOW);
>  }
> ```

So, now I finished my ideas to make it work. any idea will be welcome :blush:
thanks in advance

xreef, 2021-10-08:
Azz... sorry I lost this issue, I don't know if you fix the problem, but first of all enable debug and connect AUX pin to have more information.
Sorry again Renzo

elpamyelhsa, 2021-11-14:
I have been using this library without issues, can you confirm your code uses something like below.
```
LoRa_E32 e32ttl(&Serial2, E32_PIN_AUX, E32_PIN_M0, E32_PIN_M1, UART_BPS_RATE_9600);
[DO E32 CONFIG AS PER THE NEXT CODE BLOCK]
e32ttl.begin();
e32ttl.setMode(MODE_0_NORMAL);
[LOOP] if (e32ttl.available() > 0)
ResponseContainer rs = e32ttl.receiveMessageUntil('\n');
String message = rs.data;
```

Can you also confirm the TX>RX and RX>TX Pins are connected the right way around? To test if the E32 is responding I and reading its config on boot and check it is what I had set it too already.

```
ResponseStructContainer c;
c = e32ttl.getConfiguration();
Configuration configuration = *(Configuration *)c.data;

if (configuration.ADDL == MODULE_ADDL && configuration.ADDH == MODULE_ADDH && configuration.CHAN == MODULE_CHAN && configuration.OPTION.transmissionPower == RADIO_POWER && configuration.OPTION.wirelessWakeupTime == RADIO_WAKE_UP)
  {
    Serial.println("Loaded E32 configuration from module.");
  }
  else
  {
    Serial.println("Saving new E32 configuration to module.");
    configuration.ADDL = MODULE_ADDL;
    configuration.ADDH = MODULE_ADDH;
    configuration.CHAN = MODULE_CHAN;

    configuration.OPTION.fixedTransmission = FT_FIXED_TRANSMISSION; 
    configuration.OPTION.wirelessWakeupTime = RADIO_WAKE_UP;
    configuration.OPTION.fec = FEC_1_ON;
    configuration.OPTION.ioDriveMode = IO_D_MODE_PUSH_PULLS_PULL_UPS;
    configuration.OPTION.transmissionPower = RADIO_POWER;

    configuration.SPED.airDataRate = AIR_DATA_RATE_001_12;
    configuration.SPED.uartBaudRate = UART_BPS_9600;
    configuration.SPED.uartParity = MODE_00_8N1;
    e32ttl.setMode(MODE_3_PROGRAM);
    e32ttl.setConfiguration(configuration, WRITE_CFG_PWR_DWN_SAVE);
    e32ttl.setMode(MODE_0_NORMAL);
  }

  radio_print_parameters(configuration);
  c.close();
```

xreef, 2021-11-24:
@elpamyelhsa answer and my confirmation close the issue.
Thanks to @elpamyelhsa

closed 2021-11-24 by xreef

RaspiGame, 2021-12-13:
Hello, I wasn't testing it in a time.... 

thank you all for the answer @xreef and @elpamyelhsa :-)

I've been testing using the code-exemple "SendReceiveTransparentTransmissionE32_TTL_1W" (and all of others, ofcorse.... controlling all pins AUX, M0, M1 by the arduino).... _With that, I suppose it answer your questions, isn't it?_

just **changing** the ADDL, ADDH,  and CHAN as you can see below

and **my results** has been that: (w/o any  interection between each one)
[image]

**This is the pinout used**: LoRa_E32 e32ttl100(3, 2, 5, 7, 6); (and switching RX and TX to skip this issue)

- Do you see something strange weird, unusual...?
- Could be chip damaged?

Thank you alot again

xreef, 2021-12-14:
Hi @RaspiGame,
for transparent transmission you must set the same ADDH ADDL and Channel.
Bye Renzo

elpamyelhsa, 2021-12-16:
@RaspiGame You need to use sendFixedMessage to send messages from a different ADDL, ADDH. CHAN of the transmitter. When you send a message you will need to specify the receivers ADDL, ADDH, CHAN. If they are the same, you can then omit these settings and use Transparent Transmission.

```
ResponseStatus rs;
rs = e32ttl.sendFixedMessage(RADIO_ADDH, RADIO_ADDL, RADIO_CHAN, "HELLO WORLD");
```

Also note, it is best practice to include a radio check to test the RX and TX pins as per the code sample above, it will also limit how many times the flash memory is written to the E32 module as to not wear it out. The output should only show "Saving new E32 configuration to module." if it is a config change and on first boot. Every boot after that will show "Loaded E32 configuration from module." to prove the module is responding. The message "Loaded E32 configuration from module." on every boot will imply a wiring fault.

RaspiGame, 2021-12-19:
Hi,

I've tried with example: "sendReceiveFixedMessageStructure"

and this is the result: (nothing received)
[image]

- the issue could be different "SpeedParityBit", "SpeedUARTDatte" and "SpeedAirDataRate"? How can I fixe the value?
- What "success" message means? reception from E32

xreef, 2021-12-19:
Hi @RaspiGame,
You must set the same speed as air, baud, and the other parameter.
The configurations are different between the two devices.
Bye Renzo

RaspiGame, 2022-03-06:
### I've been working in it, creating a pcb to work easily.
[image]
with this schematic in RF part (connected to Atmel328 micro)
[image]
***
### Running "receiveFixedTransmissionStruct" and "sendFixedTransmissionStruct" examples 
with following changes:
- the pinout for "*LoRa_E32 e32ttl(20, 21, 5, 6, 7);*"
- baud rate for both codes: "*configuration.SPED.uartBaudRate = UART_BPS_115200;*" (due it was different between both)
[image]

As you can see all parameters are identical (*excepte ADDL, ADDH, CHAN). But it don't work sending message. Why "HEAD" is 0 0 0 if the communication works?

***
### Running "sendReceiveTransparentTransmissionE32_1W"
[image]
As you can see all parameters are identical (*include ADDL, ADDH, CHAN). But still don't work sending. Some time they read the "HEAD" and other time not, why? Could be a signal of device damage?

***
### enable debug
as @xreef said: 
"first of all enable debug and connect AUX pin to have more information."

How can I do that by arduino? Or I need USB-serial?

Thanks in advance

RaspiGame, 2022-03-06:
The antena could be the problem?
I'm using this one: https://es.aliexpress.com/item/4000124517198.html?gatewayAdapt=glo2esp&spm=a2g0o.9042311.0.0.274263c0sXNztG
instead one of 433MHz   ':)

xreef, 2022-03-07:
Hi @RaspiGame,
very beautiful.

to have more debug information, you must uncomment the define on the LoRa_E32.h file
`
// Uncomment to enable printing out nice debug messages.
// #define LoRa_E32_DEBUG
`

I think the antenna isn't so important, but If the device stays too close sometimes causes a problem.

Bye Renzo

xreef, 2022-03-07:
Hi @RaspiGame,
I think you must put 5v (with a good amperage) to the VCC pin of LoRa E32.
And use max power 30db.

But the device seems not to work, the first read of the configuration returns wrong data.

Try with more power, and write configuration only one time, but write the reader configuration every time.

Bye Renzo

xreef, 2022-03-07:
And delete line 550 that gives the error. I will release the fix soon.

RaspiGame, 2022-03-07:
what a mistake!!! it works!!! 

now, with this pcb boards, was a rx tx configuration issue ':) 

Thanks for your support. I invite you a beer!! Enjoy it

PS: fix issue confirmed ;)

xreef, 2022-03-08:
Hehehe.. good.
If you want share your projects/ideas I host them on my site.
Bye Renzo

RaspiGame, 2022-03-08:
I'll keep it in mind when I finish the project!