xreef/LoRa_E32_Series_Library #37: How to use listen()

fixajteknik, 2022-01-26:
Well I want to use 2 software serial in a example. but I cant use listen() commend it gives error. also if ı check listen the library it look empty method at https://github.com/xreef/LoRa_E32_Series_Library/blob/master/LoRa_E32.h#L298

xreef, 2022-01-26:
Hi @fixajteknik,
It's a required method of stream interface implementation but never used in the library. You can ignore It.
Bye Renzo

closed 2022-01-26 by xreef

fixajteknik, 2022-01-26:
So how to use 2 software serial with your library

xreef, 2022-01-26:
Ahh! sorry I don't answer your.

You can use some thing similar with explicit declaration like so:

`
SoftwareSerial mySerial(D2, D3); // Arduino RX <-- e220 TX, Arduino TX --> e220 RX
LoRa_E220 e220ttl(&mySerial, D5, D7, D6); // AUX M0 M1

SoftwareSerial mySerial2(D20, D30); // Arduino RX <-- e220 TX, Arduino TX --> e220 RX
LoRa_E220 e220ttl2(&mySerial2, D50, D70, D60); // AUX M0 M1
`

Bye Renzo