xreef/LoRa_E32_Series_Library #38: Can't get the library reliably to work with ESP32

ezcGman, 2022-02-03:
Hey there,

first, my environment:

- E32-900T20D
- Arduino IDE
- Generic ESP-32 Devboard
- E32 powered with 5V
- E32-RX wired to IO17, with a 4.7kR pull-up to 5V
- E32-TX wired to IO16, with a 4.7kR pull-up to 5V
- AUX unconnected
- M0 & M1 connected to GND

I spent a day now to get the library to work, but I just can't get it to work reliably. The main issue is, that none of your constructor examples work for me where you explicitly specify the RX/TX pins:
```
lora-test-e32-dumper:14:26: error: invalid conversion from 'int' to 'HardwareSerial*' [-fpermissive]
 LoRa_E32 e32ttl100(16, 17);  // e32 TX e32 RX
                          ^
In file included from [...]\lora-test-e32-dumper.ino:2:0:
[...]\EByte_LoRa_E32_library/LoRa_E32.h:192:3: note:   initializing argument 1 of 'LoRa_E32::LoRa_E32(HardwareSerial*, byte, UART_BPS_RATE)'
   LoRa_E32(HardwareSerial* serial, byte auxPin, UART_BPS_RATE bpsRate = UART_BPS_RATE_9600);
```
It seems that it never finds the special constructors for ESP32... I even tried adding `#define ESP32`, but no change.
The only constructors that work for me is the tree defined here: https://github.com/xreef/LoRa_E32_Series_Library/blob/master/LoRa_E32.h#L191

But I'm not sure how to use them, I don't understand your examples here, as I think they're mixed (HardwareSerial vs. SoftwareSerial): https://www.mischianti.org/2019/10/21/lora-e32-device-for-arduino-esp32-or-esp8266-library-part-2/

I googled a bit more and I got a semi working setup, looking like this:

```
#include "Arduino.h"
#include "LoRa_E32.h"

#define OPERATING_FREQUENCY 862
#define RXD2 16
#define TXD2 17

LoRa_E32 e32ttl100(&Serial2);
bool loraInitDone = false;

void setup() 
{
  Serial.begin(115200);

  while (!Serial);
  Serial.println("LoRa Sender");

  if (!e32ttl100.begin()) {
    Serial.println("Starting LoRa failed!");

    Serial.print("Rebooting...\n");
    ESP.restart();
  }
  else {
    Serial.println("Starting LoRa success!");

    loraInitDone = true;
  }
}
 
void loop() 
{
  if (loraInitDone) {
    Serial.print("Sending packet: ");
 
    ResponseStatus rs = e32ttl100.sendMessage("Prova");
    Serial.println(rs.getResponseDescription());
  }
 
  delay(5000);
}
```
This at least doesn't throw any error and it attempts to send a message, but I can't receive it on another module I have (RFM95W). So I wanted to check the configuration of the E32 module, so I used your example here: https://github.com/xreef/LoRa_E32_Series_Library/blob/master/examples/arduinoGetConfiguration/arduinoGetConfiguration.ino
I only replaced creation of the LoRa_E32 instance with my call I posted above:
```
#define RXD2 16
#define TXD2 17

LoRa_E32 e32ttl100(&Serial2);
```

And this throws an error and crashed when getting the module information:
```
No response from device! (Check wiring)
12
----------------------------------------
HEAD BIN: 11010101 213 D5
 
AddH BIN: 10111100
AddL BIN: 1101
Chan BIN: 154 -> 564MHz
 
SpeedParityBit BIN    : 1 -> 8O1
SpeedUARTDataRate BIN : 0 -> 1200bps
SpeedAirDataRate BIN  : 0 -> 0.3kbps
OptionTrans BIN       : 1 -> Fixed transmission (first three bytes can be used as high/low address and channel)
OptionPullup BIN      : 1 -> TXD, RXD, AUX are push-pulls/pull-ups
OptionWakeup BIN      : 0 -> 250ms (default)
OptionFEC BIN         : 0 -> Turn off Forward Error Correction Switch
OptionPower BIN       : 1 -> 17dBm
----------------------------------------
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x400d13b9  PS      : 0x00060530  A0      : 0x800d2249  A1      : 0x3ffb1f50  
A2      : 0x400d0000  A3      : 0x3ffbfeb8  A4      : 0x3ffb8668  A5      : 0x3ffc015c  
A6      : 0x3ffbfe48  A7      : 0x0000000c  A8      : 0x800d13b9  A9      : 0x3ffb1f20  
A10     : 0x00000000  A11     : 0x0000000c  A12     : 0x0000000c  A13     : 0x0000000c  
A14     : 0x3f400275  A15     : 0x3f400714  SAR     : 0x0000000a  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000001  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff  

ELF file SHA256: 0000000000000000

Backtrace: 0x400d13b9:0x3ffb1f50 0x400d2246:0x3ffb1fb0 0x40086189:0x3ffb1fd0

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5856
entry 0x400806a8
```

I'm lost. Do you have any clue what's totally going wrong for me?

Thanks and greetings,

Andy!

xreef, 2022-02-03:
Hi @ezcGman,
I think you must read this section of the tutorial.
https://www.mischianti.org/2019/10/21/lora-e32-device-for-arduino-esp32-or-esp8266-library-part-2/#Constructor
To change the default Serial pin of esp32 exists a specified series of constructors, you must set the pins before the Serial declaration.
Bye Renzo

ezcGman, 2022-02-03:
Hey Renzo,

thanks for the fast reply!

I'm not sure if I get what you mean. I walked though that tutorial and as I said above, these constructors don't work for me, Arduino IDE says it can't find this constructor, please see the error above. But I don't understand why. Your library only enabled these constructors if `HARDWARE_SERIAL_SELECTABLE_PIN` is defined and that is defined, if `ESP32` is defined. But Arduino IDE never seems to find this constructor and throws the error above. Even if I define `ESP32` myself...

I've set Arduino IDE to use the `DOIT ESP32 DEVKIT V1` board from the `ESP32 Arduino` board library...

Or what were you trying to say?

Thanks again and greetings,

Andy!

xreef, 2022-02-03:
Hi @ezcGman,
We'll try to find the problem.
Create a topic in the forum https://www.mischianti.org/forums/forum/mischiantis-libraries/ebyte-lora-e32-uart-devices/ and put your code there. If it's possible, place the connection schema also.
Bye Renzo

ezcGman, 2022-02-04:
Ok, let's ignore the constructor / Arduino IDE issue for now. I checked all your tutorials, incl. this one now: https://www.mischianti.org/2021/04/14/ebyte-lora-e32-device-for-arduino-esp32-or-esp8266-wor-wake-on-radio-and-new-esp32-shield-8/

I did the wiring **exactly** as you did there, even did pull ups to 3v3 instead of 5v. So it's 1:1 wired as you've shown here:
[image]

I copied the code from this example: https://github.com/xreef/LoRa_E32_Series_Library/blob/master/examples/arduinoGetConfiguration/arduinoGetConfiguration.ino
And made three changes:

1. In line 19, I swapped this `LoRa_E32 e32ttl100(2, 3); // Arduino RX <-- e32 TX, Arduino TX --> e32 RX` with line 46 from the tutorial I linked above: `LoRa_E32 e32ttl(&Serial2, 15, 21, 19); //  RX AUX M0 M1`
2. I fixed this line to set AUX to 18 instead of 15, as in the image you wire AUX to 18, not to 15!
3. I added `#define FREQUENCY_868` as the very first line in the sketch.

It compiles and **again** I partially get errors, saying "No response from device! (Check wiring)", and then crashes / guru meditates when trying to get the module information:
```
No response from device! (Check wiring)
12
----------------------------------------
HEAD BIN: 11011101 221 DD
 
AddH BIN: 10111100
AddL BIN: 1101
Chan BIN: 186 -> 1048MHz
 
SpeedParityBit BIN    : 1 -> 8O1
SpeedUARTDataRate BIN : 0 -> 1200bps
SpeedAirDataRate BIN  : 0 -> 0.3kbps
OptionTrans BIN       : 1 -> Fixed transmission (first three bytes can be used as high/low address and channel)
OptionPullup BIN      : 1 -> TXD, RXD, AUX are push-pulls/pull-ups
OptionWakeup BIN      : 0 -> 250ms (default)
OptionFEC BIN         : 1 -> Turn on Forward Error Correction Switch (Default)
OptionPower BIN       : 1 -> 17dBm
----------------------------------------
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x400d13b9  PS      : 0x00060530  A0      : 0x800d224d  A1      : 0x3ffb1f50  
A2      : 0x400d0000  A3      : 0x3ffbfeb8  A4      : 0x3ffb8668  A5      : 0x3ffc015c  
A6      : 0x3ffbfe48  A7      : 0x0000000c  A8      : 0x800d13b9  A9      : 0x3ffb1f20  
A10     : 0x00000000  A11     : 0x0000000c  A12     : 0x0000000c  A13     : 0x0000000c  
A14     : 0x3f400275  A15     : 0x3f400714  SAR     : 0x0000000a  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000001  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff  

ELF file SHA256: 0000000000000000

Backtrace: 0x400d13b9:0x3ffb1f50 0x400d224a:0x3ffb1fb0 0x40086189:0x3ffb1fd0

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5856
entry 0x400806a8
```

Also, please pay attention to the freuqency it says: `Chan BIN: 186 -> 1048MHz`. 1048MHz!? I set it to `FREQUENCY_868`

I'm seriously lost what I'm doing wrong and really looking for guidance here...

Thanks and greetings,

Andy!

xreef, 2022-02-04:
Hi @ezcGman,
I think you can try to follow these steps:
https://www.mischianti.org/forums/topic/lora_e32-h-with-esp32/
https://www.mischianti.org/forums/topic/ebyte-e32-868t30d-reboot-loop/
Bye Renzo

ezcGman, 2022-02-04:
I man, I think I just got it! Issue was power consumption! I was powering the ESP through USB and the module through the ESP and that probably way too much.

I connected the module to a separate 5V PSU, ESP still via USB (so I can debug) and shared GNDs between the ESP and the 5V PSU/E32. And SUCCESS:

```
Success
1
----------------------------------------
HEAD BIN: 11000000 192 C0
 
AddH BIN: 0
AddL BIN: 0
Chan BIN: 6 -> 868MHz
 
SpeedParityBit BIN    : 0 -> 8N1 (Default)
SpeedUARTDataRate BIN : 11 -> 9600bps (default)
SpeedAirDataRate BIN  : 10 -> 2.4kbps (default)
OptionTrans BIN       : 0 -> Transparent transmission (default)
OptionPullup BIN      : 1 -> TXD, RXD, AUX are push-pulls/pull-ups
OptionWakeup BIN      : 0 -> 250ms (default)
OptionFEC BIN         : 1 -> Turn on Forward Error Correction Switch (Default)
OptionPower BIN       : 0 -> 20dBm (Default)
----------------------------------------
Success
1
----------------------------------------
HEAD BIN: 11000011 195 C3
Freq.: 0
Version  : 45
Features : A
----------------------------------------
```

I still gonna do more tests of actual sending / reciving stuff now and read up on the links you posted later tonight!

Thanks already!

xreef, 2022-02-04:
Perfect, thanks for the feedback Renzo

closed 2022-02-04 by xreef