xreef/LoRa_E32_Series_Library #40: NANO 33 BLE support

ratio-x, 2022-02-04:
Thank you for your great work on this library. I read all the documentation. learned a lot about this module, before trying to communiacte with an E32 module (E32-868T30D) via NANO 33 BLE today. This MCU has hardware serial - dedicated TX/RX lines (D0/D1 pin) adressable via `Serial1`.

I can see there is a constructor method where I can give a reference to harware serial (Serial1) as parameter. However, compilation fails at `#include "LoRa_E32.h"`, because there is no SoftwareSerial available for this MCU.

Does this mean that this library is not usable with NANO 33 BLE?

xreef, 2022-02-04:
Hi ratio-x,
the problem It's only on this line
https://github.com/xreef/LoRa_E32_Series_Library/blob/master/LoRa_E32.h#L37

change that line with this one
`#if !defined(__STM32F1__) && !defined(ESP32) && !defined(ARDUINO_ARCH_SAMD) &&!defined(ARDUINO_ARCH_MBED)
`

I think It's the only needed change.
Give me feedback, then I release It.

Bye Renzo

ratio-x, 2022-02-05:
Hi Renzo,

thank you, it compiles now without errors. Support for NANO 33 BLE is a real enhancement.

During compilation there are many warnings. It would be great if you could fix this too. Thank you!

Mario

```
In file included from c:\data\somepath\Arduino\libraries\EByte_LoRa_E32_library/LoRa_E32.h:48:0,
                 from c:\data\somepath\Arduino\LoRa_E32_config_get\LoRa_E32_config_get.ino:17:
c:\data\somepath\Arduino\libraries\EByte_LoRa_E32_library/includes/statesNaming.h:88:17: warning: 'B00' is deprecated: use 0b00 instead [-Wdeprecated-declarations]
   MODE_00_8N1 = B00,
                 ^~~
In file included from C:\Users\someuser\AppData\Local\Arduino15\packages\arduino\hardware\mbed_nano\2.6.1\cores\arduino/api/ArduinoAPI.h:26:0,
                 from C:\Users\someuser\AppData\Local\Arduino15\packages\arduino\hardware\mbed_nano\2.6.1\cores\arduino/Arduino.h:27,
                 from C:\Users\someuser\AppData\Local\Temp\arduino-sketch-9D58A9ED2E5ECD2BBB5DBB98AF50E1FE\sketch\LoRa_E32_config_get.ino.cpp:1:
C:\Users\someuser\AppData\Local\Arduino15\packages\arduino\hardware\mbed_nano\2.6.1\cores\arduino/api/Binary.h:39:3: note: declared here
   B00       DEPRECATED(0b00      ) = 0,
   ^~~
In file included from c:\data\somepath\Arduino\libraries\EByte_LoRa_E32_library/LoRa_E32.h:48:0,
                 from c:\data\somepath\Arduino\LoRa_E32_config_get\LoRa_E32_config_get.ino:17:
c:\data\somepath\Arduino\libraries\EByte_LoRa_E32_library/includes/statesNaming.h:89:17: warning: 'B01' is deprecated: use 0b01 instead [-Wdeprecated-declarations]
   MODE_01_8O1 = B01,
                 ^~~
In file included from C:\Users\someuser\AppData\Local\Arduino15\packages\arduino\hardware\mbed_nano\2.6.1\cores\arduino/api/ArduinoAPI.h:26:0,
                 from C:\Users\someuser\AppData\Local\Arduino15\packages\arduino\hardware\mbed_nano\2.6.1\cores\arduino/Arduino.h:27,
                 from C:\Users\someuser\AppData\Local\Temp\arduino-sketch-9D58A9ED2E5ECD2BBB5DBB98AF50E1FE\sketch\LoRa_E32_config_get.ino.cpp:1:
C:\Users\someuser\AppData\Local\Arduino15\packages\arduino\hardware\mbed_nano\2.6.1\cores\arduino/api/Binary.h:47:3: note: declared here
   B01       DEPRECATED(0b01      ) = 1,
   ^~~
In file included from c:\data\somepath\Arduino\libraries\EByte_LoRa_E32_library/LoRa_E32.h:48:0,
                 from c:\data\somepath\Arduino\LoRa_E32_config_get\LoRa_E32_config_get.ino:17:
c:\data\somepath\Arduino\libraries\EByte_LoRa_E32_library/includes/statesNaming.h:90:17: warning: 'B10' is deprecated: use 0b10 instead [-Wdeprecated-declarations]
   MODE_10_8E1 = B10,
                 ^~~
In file included from C:\Users\someuser\AppData\Local\Arduino15\packages\arduino\hardware\mbed_nano\2.6.1\cores\arduino/api/ArduinoAPI.h:26:0,
                 from C:\Users\someuser\AppData\Local\Arduino15\packages\arduino\hardware\mbed_nano\2.6.1\cores\arduino/Arduino.h:27,
                 from C:\Users\someuser\AppData\Local\Temp\arduino-sketch-9D58A9ED2E5ECD2BBB5DBB98AF50E1FE\sketch\LoRa_E32_config_get.ino.cpp:1:
C:\Users\someuser\AppData\Local\Arduino15\packages\arduino\hardware\mbed_nano\2.6.1\cores\arduino/api/Binary.h:54:3: note: declared here
   B10       DEPRECATED(0b10      ) = 2,
   ^~~
In file included from c:\data\somepath\Arduino\libraries\EByte_LoRa_E32_library/LoRa_E32.h:48:0,
                 from c:\data\somepath\Arduino\LoRa_E32_config_get\LoRa_E32_config_get.ino:17:
c:\data\somepath\Arduino\libraries\EByte_LoRa_E32_library/includes/statesNaming.h:91:17: warning: 'B11' is deprecated: use 0b11 instead [-Wdeprecated-declarations]
   MODE_11_8N1 = B11
                 ^~~
In file included from C:\Users\someuser\AppData\Local\Arduino15\packages\arduino\hardware\mbed_nano\2.6.1\cores\arduino/api/ArduinoAPI.h:26:0,
                 from C:\Users\someuser\AppData\Local\Arduino15\packages\arduino\hardware\mbed_nano\2.6.1\cores\arduino/Arduino.h:27,
                 from C:\Users\someuser\AppData\Local\Temp\arduino-sketch-9D58A9ED2E5ECD2BBB5DBB98AF50E1FE\sketch\LoRa_E32_config_get.ino.cpp:1:
C:\Users\someuser\AppData\Local\Arduino15\packages\arduino\hardware\mbed_nano\2.6.1\cores\arduino/api/Binary.h:61:3: note: declared here
   B11       DEPRECATED(0b11      ) = 3,
   ^~~
In file included from c:\data\somepath\Arduino\libraries\EByte_LoRa_E32_library/LoRa_E32.h:48:0,
                 from c:\data\somepath\Arduino\LoRa_E32_config_get\LoRa_E32_config_get.ino:17:
c:\data\somepath\Arduino\libraries\EByte_LoRa_E32_library/includes/statesNaming.h:116:19: warning: 'B000' is deprecated: use 0b000 instead [-Wdeprecated-declarations]
   UART_BPS_1200 = B000,
                   ^~~~
In file included from C:\Users\someuser\AppData\Local\Arduino15\packages\arduino\hardware\mbed_nano\2.6.1\cores\arduino/api/ArduinoAPI.h:26:0,
                 from C:\Users\someuser\AppData\Local\Arduino15\packages\arduino\hardware\mbed_nano\2.6.1\cores\arduino/Arduino.h:27,
                 from C:\Users\someuser\AppData\Local\Temp\arduino-sketch-9D58A9ED2E5ECD2BBB5DBB98AF50E1FE\sketch\LoRa_E32_config_get.ino.cpp:1:
C:\Users\someuser\AppData\Local\Arduino15\packages\arduino\hardware\mbed_nano\2.6.1\cores\arduino/api/Binary.h:40:3: note: declared here
   B000      DEPRECATED(0b000     ) = 0,
   ^~~~
In file included from c:\data\somepath\Arduino\libraries\EByte_LoRa_E32_library/LoRa_E32.h:48:0,
                 from c:\data\somepath\Arduino\LoRa_E32_config_get\LoRa_E32_config_get.ino:17:
[... cut ...]
                 from C:\Users\someuser\AppData\Local\Temp\arduino-sketch-9D58A9ED2E5ECD2BBB5DBB98AF50E1FE\sketch\LoRa_E32_config_get.ino.cpp:1:
C:\Users\someuser\AppData\Local\Arduino15\packages\arduino\hardware\mbed_nano\2.6.1\cores\arduino/api/Binary.h:61:3: note: declared here
   B11       DEPRECATED(0b11      ) = 3,
   ^~~
Sketch uses 88536 bytes (9%) of program storage space. Maximum is 983040 bytes.
Global variables use 43944 bytes (16%) of dynamic memory, leaving 218200 bytes for local variables. Maximum is 262144 bytes.

--------------------------
Compilation complete.

```

xreef, 2022-02-06:
Thanks again, I'll release a version soon to remove the warnings and add the BLE support.
Bye Renzo

xreef, 2022-02-23:
Version released.
Thanks for the support

closed 2022-02-23 by xreef