xreef/LoRa_E32_Series_Library #43: Raspberry Pi Pico with LoRa E32 library

janczeresnia, 2022-04-07:
Hi
You wrote a very nice library. Is there support for Raspberry Pi Pico microcontroller? And if so, how to configure it?
Library works great with Seed Studio XIAO and ESP8266 D1 but not with Raspberry Pi Pico :(

xreef, 2022-04-07:
Hi @janczeresnia,
I'll start work on Pi Pico and STM32 in the next few days. 
I'm going also to create new shields for that devices.

Bye Renzo

janczeresnia, 2022-04-07:
Thank you for the quick reply. Regards.

xreef, 2022-04-08:
I reopen this issue, and I close It when development will be effective, so I alert you.
Bye Renzo

IModThings, 2022-08-11:
Any update on this?

xreef, 2022-08-11:
Hi,
I must test it, I think probably it work, but not yet tested.
If I find some time I test it in the next week.
Bye Renzo

xreef, 2022-08-16:
Hi all,
I tested the Raspberry Pi Pico and It's fully supported.

You can find here the configuration and wiring diagram.

https://www.mischianti.org/forums/topic/ebyte-lora-uart-arduino-library-raspberry-pi-pico-support/

Bye Renzo

closed 2022-08-16 by xreef

IModThings, 2022-08-16:
Tysm!