xreef/LoRa_E32_Series_Library #46: ESP32 & SoftwareSerial error

nravanelli, 2022-07-18:
I seem to be having an issue with the following code which mirrors examples:

```
#include <SoftwareSerial.h>
// LORA communication
#include "LoRa_E32.h"
//------ SOFTWARE SERIAL 
#define RX_PIN 19                                          // Rx pin 
#define TX_PIN 23                                          // Tx pin                                    
SoftwareSerial mySerial(RX_PIN, TX_PIN);                   

//-------------LORA E32 connection
LoRa_E32 e32ttl(&mySerial);
```

and I am met with this error:
`no matching function for call to 'LoRa_E32::LoRa_E32(SoftwareSerial*)'`

Based on the constructors tutorials provided, this should work, unless there is something different with an esp32? ... https://www.mischianti.org/2019/10/21/lora-e32-device-for-arduino-esp32-or-esp8266-library-part-2/#Constructor

xreef, 2022-07-19:
Hi nravanelli,
[by default, I disable SoftwareSerial for ESP32](https://github.com/xreef/LoRa_E32_Series_Library/blob/master/LoRa_E32.h#L44) because [ESP32 allows selecting pins to assign to his HardwareSerial](https://www.mischianti.org/2019/10/21/lora-e32-device-for-arduino-esp32-or-esp8266-library-part-2/#Constructor).
If you want to enable that functionality, you must add` #define ACTIVATE_SOFTWARE_SERIAL `before the use of It.
Bye Renzo

xreef, 2022-07-20:
Put the #define ACTIVATE_SOFTWARE_SERIAL inside LoRa_E32.h file, probably the value isn't propagated in the other files.
Bye Renzo

closed 2022-07-25 by xreef

ikidd, 2024-09-02:
I beat my head against this issue for 2 days before finding this issue to get my code to compile.  

It would be awesome if the documentation made mention of it, because the `if !defined ` section to enable that in Lora_E32.h doesn't work with my ESP32 model chosen for compile for some reason.