xreef/LoRa_E32_Series_Library #49: How to configuration set long distance...?

nthink-hoon, 2022-11-03:
[image]

I tried changing the air data rate setting, but the 1:1 communication distance from the outside is less than 10~15m,
and communication is not possible over the 15m distance.

If i set the transmission cycle to 7 seconds, it sometimes takes 10 to 15 seconds to receive within 15 meters.
What should I do?

cattHoons, 2022-11-04:
+) i use ES32-900T30D

xreef, 2022-12-14:
Hi,
sorry I lost this message, but to get a good distance you need a good amperage and a big antenna.
https://www.mischianti.org/forums/topic/e22-distance-test/
Bye Renzo

closed 2022-12-14 by xreef