xreef/LoRa_E32_Series_Library #55: Using E32-433T33D full power (33dBm)

Nuno-Miguel, 2023-08-08:
Hello. I am starting to use Renzo’s library with two E32 LoRa modules and two Arduinos. So far thing are going well thanks to Renzo Mischianti tutorials and library (both great). I am using transparent transmission and 0.3kbps airdate rate, hoping to reach 10 km transmission distance (at continuous operation). To achieve this I am using an E32-433T33D as permanent TRANSMISSOR (maximum power 33dBm) and an E32-433T20D (maximum power 20dBm) as permanent RECEIVER. I am powering the E32-433T20D from Arduino 5V pin, and using an external 5V power supply able to provide 2.4A to the E32-433T33D. Both LoRa modules were connect using the proposed “LoRa E32 TTL 100 Arduino fully connected” schematic given by Renzo at tutorial number 2. I was able to put both LoRa modules communications using the examples given by Renzo! :)

In this context, I have two questions I would like to place. Please attend that my knowledge of Arduino is low, so forgive me if they are dummy.

1. I am unable to use the full power of the E32-433T33D… Using #define E32_TTL_1W I am using 30dbm (I think), but how can I use the full 33dBm of the LoRa module?
2. Can the code be more efficient? What should I change to make continuous transparent transmission more efficient?

Regards
Nuno

RECEIVER CODE:

`// LoRa Module: E32-433T20D
// M0: PIN D6, 3.3v (ON) to set configuration, 0.0v (OFF) to comunicate.
// M1: Pin D5, 3.3v (ON) to set configuration, 0.0v (OFF) to comunicate.
// TX: PIN D2
// RX: PIN D3
// AUX: Not used.
// VCC: 5v (external power supply must make sure the current output ability is more than 250mA and ensure the power supply ripple within 100mV)
// GND: GND

#define E32_TTL_100 // Module LoRa: E32-433T20D

#include "Arduino.h"
#include "LoRa_E32.h"

LoRa_E32 e32ttl100(2, 3);
void printParameters(struct Configuration configuration);
void printModuleInformation(struct ModuleInformation moduleInformation);

void setup() {

  pinMode(6, OUTPUT); // M0
  pinMode(5, OUTPUT); // M1
  digitalWrite(6, HIGH); // M0 = 3.3V
  digitalWrite(5, HIGH); // M1 = 3.3V
  
  Serial.begin(115200);
	delay(500);

	e32ttl100.begin();

	ResponseStructContainer c;
	c = e32ttl100.getConfiguration();
	Configuration configuration = *(Configuration*) c.data;
	Serial.println(c.status.getResponseDescription());
	Serial.println(c.status.code);

  // LoRa module configuration.
	printParameters(configuration);
	configuration.ADDL = 0x0;
	configuration.ADDH = 0x1;
	configuration.CHAN = 0x17;
	configuration.OPTION.fec = FEC_1_ON;
	configuration.OPTION.fixedTransmission = FT_TRANSPARENT_TRANSMISSION;
	configuration.OPTION.ioDriveMode = IO_D_MODE_PUSH_PULLS_PULL_UPS;
	configuration.OPTION.transmissionPower = POWER_20;
	configuration.OPTION.wirelessWakeupTime = WAKE_UP_250;
	configuration.SPED.airDataRate = AIR_DATA_RATE_000_03;
	configuration.SPED.uartBaudRate = UART_BPS_9600;
	configuration.SPED.uartParity = MODE_00_8N1;

	ResponseStatus rs = e32ttl100.setConfiguration(configuration, WRITE_CFG_PWR_DWN_LOSE);
	Serial.println(rs.getResponseDescription());
	Serial.println(rs.code);
	printParameters(configuration);
	c.close();

}

[... 113 lines cut ...]
  delay(10);
  
}

void printParameters(struct Configuration configuration) {
  
	Serial.println("----------------------------------------");
	Serial.print(F("HEAD : "));  Serial.print(configuration.HEAD, BIN);Serial.print(" ");Serial.print(configuration.HEAD, DEC);Serial.print(" ");Serial.println(configuration.HEAD, HEX);
	Serial.println(F(" "));
	Serial.print(F("AddH : "));  Serial.println(configuration.ADDH, BIN);
	Serial.print(F("AddL : "));  Serial.println(configuration.ADDL, BIN);
	Serial.print(F("Chan : "));  Serial.print(configuration.CHAN, DEC); Serial.print(" -> "); Serial.println(configuration.getChannelDescription());
	Serial.println(F(" "));
	Serial.print(F("SpeedParityBit     : "));  Serial.print(configuration.SPED.uartParity, BIN);Serial.print(" -> "); Serial.println(configuration.SPED.getUARTParityDescription());
	Serial.print(F("SpeedUARTDatte  : "));  Serial.print(configuration.SPED.uartBaudRate, BIN);Serial.print(" -> "); Serial.println(configuration.SPED.getUARTBaudRate());
	Serial.print(F("SpeedAirDataRate   : "));  Serial.print(configuration.SPED.airDataRate, BIN);Serial.print(" -> "); Serial.println(configuration.SPED.getAirDataRate());

	Serial.print(F("OptionTrans        : "));  Serial.print(configuration.OPTION.fixedTransmission, BIN);Serial.print(" -> "); Serial.println(configuration.OPTION.getFixedTransmissionDescription());
	Serial.print(F("OptionPullup       : "));  Serial.print(configuration.OPTION.ioDriveMode, BIN);Serial.print(" -> "); Serial.println(configuration.OPTION.getIODroveModeDescription());
	Serial.print(F("OptionWakeup       : "));  Serial.print(configuration.OPTION.wirelessWakeupTime, BIN);Serial.print(" -> "); Serial.println(configuration.OPTION.getWirelessWakeUPTimeDescription());
	Serial.print(F("OptionFEC          : "));  Serial.print(configuration.OPTION.fec, BIN);Serial.print(" -> "); Serial.println(configuration.OPTION.getFECDescription());
	Serial.print(F("OptionPower        : "));  Serial.print(configuration.OPTION.transmissionPower, BIN);Serial.print(" -> "); Serial.println(configuration.OPTION.getTransmissionPowerDescription());
	Serial.println("----------------------------------------");

}`

SurenBono, 2023-11-17:
... why did you not setup the module with the . exe and just run simpler code on the module..

xreef, 2023-11-17:
Hi,
the 33 is managed by E32_TTL_2W define.
For the transparent transmission you must consider that you can't send a packet size major than the size configured, so pay attention when streaming the data.
Bye Renzo

closed 2023-11-17 by xreef

SurenBono, 2023-11-18:
... Maximum tx/rx buffer on Arduino is 64 bytes.. I assume e32 is around 58
bytes.. but I'm using less than those.. example: char Transmit [] =
"Wednesday,23.12.23,AM 10:45:34,U+8,"
// 33 bytes average

.. but I'm concerned of its durability ,can It transmit decoded GPS every
seconds for at least 5 years ... last time I tested it on hc-12 .. it broke
down after a few days of continues transmission.. not sure what went
wrong... is it limited by it's duty cycle?... so far my Gps neo-6m on led
matrix did not have this problem.. its still running smoothly after 3 year
at least ... kudos for that at least..

xreef, 2023-11-18:
Hi,
I think that It could broken if there was a  memory leak. Try to check It.
Bye Renzo

SurenBono, 2023-11-20:
... I tried 5 PCs (hc-12)of it.. same results after a few days... something
limiting the tx frequency... probably it's duty cycle.. or memory
fragmentation problem .. or something else.. I'm not sure... I cant update
it's firmware either.. it's complicated process