xreef/LoRa_E32_Series_Library #57: Can I just do simple Serial. write without this library ?

SurenBono, 2023-11-16:
example:

char[] a="tx1, ABC,transmit test,123,noise check,”;
Serial. write(a, sizeof (a)-1);

xreef, 2023-11-17:
Hi,
yes you can, but you can use only transparent transmission or manage addressing ecc. and you must do all the configuration with the library or in manual mode.
Bye Renzo

closed 2023-11-17 by xreef

SurenBono, 2023-11-17:
Hi thank you for your reply... you mean I set the parameters with the .exe
and only use SoftwareSerial. h while it does the magic..

xreef, 2023-11-17:
For transparent transmission, yes, but you must pay attention to not sending new data before finishing the previous send and checking the packet size.
Bye Renzo

SurenBono, 2023-11-18:
.. of course... I will use if data available to send function ..

void transmitData() {
if(Serial2.availableForWrite()){
Serial2.write(TX,sizeof(TX)-1);}