xreef/LoRa_E32_Series_Library #61: Monitor node with multiple responses at the same time

bpstermometria, 2024-02-03:
I have a hub node that sends a broadcast command to all nodes. It waits for responses to check whether all devices responded correctly, however, devices respond at the same time, and the ReceiveMessage method only captures the response from one device, discarding responses from all others. I would like to know what is the correct way to have a monitor node to handle multiple messages at the same time.

NOTE: The nodes are far from the monitor node.

NOTE: I didn't post the code because it's a simple example, I send a broadcast message to all devices, and all devices respond at the same time, but only the response from one device is processed.

NOTE 3: The host device is an ESP32.

xreef, 2024-02-06:
Hi BPS,
The monitor does not discard all others. It receives the message and puts It on the buffer.
The buffer is 512 bytes, and the message size is 58, so It can receive 512/58 messages without reading the buffer.
It's a best practice to read the buffer as soon as possible.
Bye Renzo

xreef, 2024-02-06:
Try to use a delimiter and use receiveMessageUntil to do a check.
Bye Renzo

closed 2024-02-14 by xreef