xreef/LoRa_E32_Series_Library #65: ESP-IDF port

alessandrv, 2024-04-30:
Hello, I tried using this library with esp-idf, but I can't seem to get it working. No error messages or anything, just can't receive or send messages (the other device using it on arduino). Do you have any experience with esp-idf and could give it a try to see if I'm missing something? Thanks for your time and the amazing library.

xreef, 2024-05-01:
Hi, 
If I remember @fixajteknik use idf with my library. 
I don't have experiences with it.
Bye Renzo

fixajteknik, 2024-05-01:
https://youtu.be/6fyVCYDoc2Q?si=buLvKkdy85iC2RQ8

Also there is playlist for idf Arduino as component for all library of Arduino

alessandrv, 2024-05-01:
> https://youtu.be/6fyVCYDoc2Q?si=buLvKkdy85iC2RQ8
> 
> Also there is playlist for idf Arduino as component for all library of Arduino

Thankk, im actually using Arduino as a compenent. It i wasn't I would be getting errors because of the libraries that this library depends on. I just noticed 2 things using the example. I'm using hardware serial port 2, when I init the serial 1 for the print the scripts hangs on the init forever. If I remove that and use printf() to printf stuff I get this error:
__digitalRead: pin (whatever pin I used as aux/m0/m1) is not set as GPIO (this is more or less the error not at the computer right now). Have you ever got this error and maybe know what's going on? Thank you in advance

closed 2024-05-16 by xreef