xreef/LoRa_E32_Series_Library #67: e32ttl.available() always returns false

adiiot-1316, 2024-06-08:
Hi Team,

Thanks for such wonderful library. I am using esp32-c6 and trying to connect with E32-900t30d-v8.

| E32-900t30d-v8. | ESP32-C6 |
| ------ | ------ |
| M0 | 10 |
| M1 | 11 |
| AUX | -1 |
| RX | 16 |
| TX | 17 |
| VCC | 5v |
| Gnd | GND |

In Loop function, e32.available()always returning false. Below is the code being used. Any help would be appreciated.

```sh
`#include "Arduino.h"
#include "LoRa_E32.h"
 
LoRa_E32 e32ttl(&Serial1, -1, 10, 11); //  RX AUX M0 M1
void setup() {
  Serial.begin(9600);
  Serial1.begin(9600);
  delay(4500);
 // Startup all pins and UART
  e32ttl.begin();
 
	
  // Send message
  ResponseStatus rs = e32ttl.sendMessage("Hello, world?");
  // Check If there is some problem of succesfully send
  Serial.println(rs.getResponseDescription());
}
 
void loop() {
    // If something available
  if (e32ttl.available()>1) {
    Serial.println("e32ttl is available ");
      // read the String message
    ResponseContainer rc = e32ttl.receiveMessage();
 
    // Is something goes wrong print error
    if (rc.status.code!=1){
        Serial.println(rc.status.getResponseDescription());
    }else{
        // Print the data received
        Serial.println(rc.status.getResponseDescription());
        Serial.println(rc.data);
    }
  }
  if (Serial.available()) {
      Serial.println("In Serial send");
      String input = Serial.readString();
      e32ttl.sendMessage(input);
  }
  delay(3000);
}`
```

satyanvesi, 2024-06-08:
are you connected pull up resistors with tx and rx pin?
what does the -1 meaning in this line LoRa_E32 e32ttl(&Serial1, -1, 10, 11)?

adiiot-1316, 2024-06-08:
@satyanvesi  : Thanks for reaching out. 

I am just directly connecting the e32 pins with  esp32. Not using pull up resistors. Can you please how to achieve that?
-1 means not using aux signal.

satyanvesi, 2024-06-08:
you should go with https://mischianti.org/lora-e32-device-for-arduino-esp32-or-esp8266-library-part-2/
this guide and follow the esp32 schematic for connecting the resistors.

adiiot-1316, 2024-06-08:
@satyanvesi  : Its still not working even after adding pull up. Only thing is e32ttl.available() is not returning >1 . Possible to connect over email(adi1316@gmail.com) please ? Its bit kind of blocker in my live project. Really appreciate your support here.

closed 2024-06-23 by adiiot-1316