xreef/LoRa_E32_Series_Library #69: Examples don't work without adding a serial.begin

ikidd, 2024-09-09:
The example sketches in the Arduino IDE library you ship don't work unless you manually start the serial port in setup() before invoking the LoRa_E32 object.

Need to add something like:

`Serial2.begin(9600,SERIAL_8N1,16,17); //16 RX 17 TX`

in order to get the object defined by:

`LoRa_E32 e32ttl(&Serial2, 21,18,19)`

like in the example sketch "esp32_e32_wake_up_from_WOR_deep_sleep"

I'm guessing from the lack of success I had on softwareserial invocations, something similiar needed to be done with the software serial device that was instantiated and passed as `&mySerial` or as just rx/tx pins, because I had no success getting a softwareserial port to work before I discovered this about using the UART2 with the library.

xreef, 2024-09-09:
Hi ikidd,
if you use core 3, the serial pins are changed, so you must specify them like so.
`LoRa_E32 e32ttl(16, 17, &Serial2, 21,18,19)`
or set the correct one.
Bye Renzp

closed 2024-09-09 by xreef

ikidd, 2024-09-09:
What's core 3? I'm using the Arduino IDE, which won't take that syntax as you've written.  Gets a constructor function call error, as I would expect looking in the library code.  

I'm using this library and examples as installed from the Arduino repo in the Library Manager.

xreef, 2024-09-10:
Hi,
sorry the constructor is like this one
`LoRa_E32(byte txE32pin, byte rxE32pin, HardwareSerial* serial, UART_BPS_RATE bpsRate = UART_BPS_RATE_9600, uint32_t serialConfig = SERIAL_8N1);`
you can find the documentation [here](https://mischianti.org/lora-e32-device-for-arduino-esp32-or-esp8266-library-part-2/#Constructor).

Here the [arduino-esp32 core ](https://github.com/espressif/arduino-esp32)you use on arduino.